opentx/opentx #4568: RC11 N363 on Taranis Q X7

JR63, 2017-03-06:
Nice effort with the RC11 N323, I like it very much!

I found a minor issue:

I can set a Logical Switch with e.g. Cells 3.25V:

[image]

But in the Logical Switches overview it looks like this:

[image]

I think it should be:
Cels 3,25V

Tested with real Q X7 and Companion.

Bye
JR

3djc, 2017-03-07:
I need more details, as I cannot reproduce it:
- is it coming for a earlier verison ?
- how was it created ?

[image]

[image]

JR63, 2017-03-07:
Hi,

I think it's one line in the following method:

void menuModelLogicalSwitches(event_t event)
{
     ...

      else {
        uint8_t v1 = cs->v1;
        drawSource(CSW_2ND_COLUMN, y, v1, 0);
        if (v1 >= MIXSRC_FIRST_TELEM) {
          drawSourceCustomValue(CSW_3RD_COLUMN, y, v1, convertLswTelemValue(cs), LEFT);
        }
        else {
          if (v1 <= MIXSRC_LAST_CH) {
            drawSourceCustomValue(CSW_3RD_COLUMN, y, v1, calc100toRESX(cs->v2), LEFT);
          }
          else {
            drawSourceCustomValue(CSW_3RD_COLUMN, y, v1, cs->v2, LEFT);
          }
        }
      }

     ...
}

The line:
        uint8_t v1 = cs->v1;

maybe shall be:
        int v1 = cs->v1;

It's the same problem as the fix of #4416

v1_val = (uint8_t)cs->v1;
->
v1_val = cs->v1;

Bye
JR

3djc, 2017-03-07:
I still would like the otx to confirm a fix if possible. Thx

JR63, 2017-03-07:
No problem, here is my ziped otx:

[document1.zip](https://github.com/opentx/opentx/files/823790/document1.zip)

3djc, 2017-03-07:
Thx, I have succeeded reproducing it, will make my fix much easier to check :)

JR63, 2017-03-07:
cool

3djc, 2017-03-07:
Your fix would work, but we have type defined for this scenario :)

JR63, 2017-03-07:
ok, source_t is also a good choice ;-)

#if defined(PCBTARANIS) || defined(PCBHORUS)
typedef uint16_t source_t;
#else
typedef uint8_t source_t;
#endif

perfect, thanks.

closed 2017-03-07 by bsongis

tonythetiger95, 2017-03-11:
How can I get this fixed exactly. Pretty new to the whole opentx platform and so far I'm having a good experience with the sound editing and so on but not really sure how to fix this issue thanks.

3djc, 2017-03-11:
When RC12 gets released, it will have the fix

tonythetiger95, 2017-03-11:
Thank you. I'll be sure to check back in when that gets put out. Good day

PR #4572 (2017-03-07): Casting fix
Thix fixes #4568