opentx/opentx #4571: Character alignment issue?

rdeanchurch, 2017-03-07:
[image]
[QX7.zip](https://github.com/opentx/opentx/files/823167/QX7.zip)

On n363 (and n362), QX7 the character alignment is not what I would have expected.  Notice the Mid and DBL size appear to be at least one pixel too low.  This is true in both the simulator in Companion and on the tx as shown in the attached picture.
Data written from a telemetry script, QX7, also attached.

3djc, 2017-03-07:
It is not X7 specific, X9 is exhibiting the same behaviour, still investigating

[image]

3djc, 2017-03-07:
ok, what happens is that the font itself has a blink pixel line in MID and DBL size;
[image]

It can be fixed, but will have impact, so needs to be assessed carefully

closed 2019-11-15 by stale[bot]