opentx/opentx #4574: Lua enhancement idea: Lua apps menu / system

RealTadango, 2017-03-07:
With S.Port and Lua great things are possible and the list of one time / configuration scripts is slowly growing.... but there is no good / nice way to start a lua from the AS. I have an idea to make it much nicer / user friendly (Mainly for Horus for now):

In the long press enter menu add a new option: Apps. This will show an app browser which shows a list of lua apps (located in SCRIPTS/APPS). When highlighting one of the app names an image can be show on the right, like when browsing images on the sd. This way a nice icon / preview of the application can be shown, or more information / about. When selecting the app it will start the same as started from the SD. The menu is just a nice shortcut.

In the main lua return new options could be added for applications scripts:
 - appname: If set this name will be shown in the list. Is missing then the filename is used
 - appimage: Path to the icon / image to be shown in the list

File structure would be SCRIPTS/APPS/<APP FOLDER>/<APP FILES>.

Current list of applications that i know:
 - PID / VTX configuration
 - S6R / S8R setup
 - LView log viewer
 - ECU Terminal

With new smart sensors more will come

3djc, 2017-03-07:
We have this need on our longer term todo list, but we might not limit to lua only, so it might take a different form

RealTadango, 2017-03-07:
Ok, as long as apps can be accessed quickly it will be great :)

projectkk2glider, 2017-03-07:
Yes we need something like this.

lshems, 2017-03-07:
You can do all this pretty easily in lua, with one script that loads the others, based on a list and select action.

Only thing is you can't get a directory listing now. So you have to either name them to something you can find, as was done with the telemetry scripts on 2.0, who where named TELEM01.lua - TELEM07.LUA. Simply scan the directory for these filenames by trying to load them.
Or you need to maintain a list somewhere to know what to load.

Off course it would then be nice if the script would return a name, next to the run, init and background functions. But there is nothing that prevents you from incorporating that info in the return of the script already.

So, in short, this can already be done, and if a "getfilelist(directory)" type of command would be added to LUA, all becomes even more simple to create without changing the Horus software at all.

RealTadango, 2017-03-07:
The main issue is starting this script. We don't want to go the SD browser and select a script to start. If you want to keep you SD organised it is to much hassle to browse folders every time you want to launch the S6R config script and switch back to the model programming.

kilrah, 2017-03-07:
We plan it to be in the native UI, maybe a "utilities" page in the radio menu.

lshems, 2017-03-07:
Horus I don't know about. Taranis and X7 is one long page press.

But it's nice to have in the GUI of OpenTX, I assume :)

azaz44, 2018-08-31:
Definitely a needed feature. Browsing SD card is good for devs but not for normal usage.

I think a smaller version of the proposed idea would be to just let users add new items to the menu, which pops up after you hold the dial (the one with Reset, Statistics, Select Model...). User could just specify list of scripts to run and names to display. This would be needed in two places - SYS and MDL, just like Special / Global functions are made. This is because some scripts are per model (heli flybarless stuff for example), and some are global.

The original idea is nice (although I'm not sure if I like an additional click to go to "apps"). But if it would take ages to implement, maybe a simple idea is better than nothing?

lshems, 2018-08-31:
Autostart a Lua script on model select.

Solves all this, and everyone is free to choose how their GUI will look from there.

Use LuA to build your custom library and GUI.

User expectations and requirements will differ hugely which complexifies a general approach.

mpaperno, 2018-08-31:
Basically sounds like this idea is a shortcut to the SD browser starting in a particular folder.  The "app name" and "app image" could be implemented for any .lua file (good idea! esp. with those 6 char file names).  They could for example just be specified in comments at the top of the file (SD browser could try to read the first 2 lines to find them), which would be faster than loading/executing each script to determine return values.

That would all be pretty easy to implement I think... there would be very little new functionality, anyway, mostly re-using what already exists.

I also have a patch ready which adds file operations for Lua scripts, including directory listing. One could build their own custom browser/loader with that.  Perhaps as an alternative to the "generic" option discussed here.

>Autostart a Lua script on model select.

How would that help execute arbitrary scripts ("apps") whenever one wishes?  Why would you want that tied to model select?  Sounds like a separate feature... ?

lshems, 2018-09-01:
Why not returning the names and other info on a script call, as is done for widgets?

Please stick with Lua interpretable coding, otherwise I would have to specify the sale name twice. Once in some header line, and one in the code for Lua use in own loaders.

Autostart would prevent browsing through the menus, but indeed can be seen as seperate feature.

mpaperno, 2018-09-01:
> Why not returning the names and other info on a script call

Like I said, much faster. Also less memory use. Also less memory use inside the actual script (no code required).  But, perhaps there would be other features required which would need actual parsing... not saying that's a bad idea, but it will use more resources.

> as is done for widgets

Because you want to preview the widget as it will actually appear in the space?  Vs. an "app" would just have a static icon.  Plus the way widgets are loaded is different anyway IIRC, the Lua is only parsed once.

> Once in some header line, and one in the code for Lua use in own loaders.

I don't know what this means ("own loaders" ?).  Having a name appear twice is a small price to pay for the performance gains IMHO, if that's even an issue for anyone besides you.

```lua
-- script name: My Fancy App
-- script icon: ./fancyapp/icon.png
```

mpaperno, 2018-09-01:
>  Plus the way widgets are loaded is different anyway

Which would of course be a whole other/different way to go for "apps," with a formally defined API, loaded into C++ objects, and so on, like is done for custom Widgets and Layouts now.  I'm not sure how feasible this will be to pull off on the non-Horus radios with limited RAM, but maybe.  It would also be more work than the simple version I was describing.  I'm not sure I see a clear benefit, at least in the short term for what was proposed/requested in the OP.

lshems, 2018-09-01:
As for the header part, if you 'return' the parameters when asked for in the start of the code, no further code is interpreted, saving time and memory.

I can read up to 30 long scripts file names using that. It is already implemented using Lua in my menu setup without problems.

If neccesary, limit the read to the list visible and add when scrolling through the list.

I pass an option variable to the script. If the option is 'I' from info, it returns an array with  info, just as a model.getfunctiion() does, with a return, si the script is not further interpreted.

External c code is not asked for I think.

Hope this helps. Just trying to prevent yet another opentx specialty, and speed is not an issue.

Actually, when thinking of it, for taranis it would help if a default script is assigned to a telemetry page, and the screen could be activated by a switch.

When starting this menu system as a telemetry script, only the Lua file functions need to be expanded for general use.

As said, I already have an app browser in place completely in Lua, it just can't scan the sdcard, only a custom directory where the scripts are put in with filenames known upfront, like menu1, menu2. These scripts are just wrappers in most cases, loading another 'standard' telemetry script. I just need them because if a missing during function in Lua.

I speak a lot of what I have in place, but anyone can really easy make a similar menu. It's not rocketscience to read a directory listing and read a header per file and show those haeders in a list to click on them for loading them.

Please don't add custom stuff if standard Lua with file listings could do it. Opentx is complex enough.

mpaperno, 2018-09-01:
> As for the header part, if you 'return' the parameters when asked for in the start of the code, no further code is interpreted

There's no such thing, or I don't understand what you're talking about.  The whole script/file needs to be loaded and parsed before anything can be returned.  Please show an example of what you mean.

> As said, I already have an app browser in place completely in Lua

Share it so we have a clue what you're talking about.

> Please don't add custom stuff if standard Lua with file listings could do it.

I don't know what you're calling "standard Lua."  There's no "standard" for "please show me your pretty name and an image" when loading a script/file.  Either way there needs to be some kind of implementation and documentation, whether it's "put these comments at the top of your file" or "your script can also return two values named 'appname' and 'appimage'."

lshems, 2018-09-01:
I have a small loader script for each menu item.

If it receives an 'I' as first passed parameter, it will return the name of the menu item, if not, it will call the associated script.

Really fast, and no need to have two special lines to read.

Also,you need to think about submenus.

In the end, I think the default file browser with a default home dir pointing to the scripts and calling it 'app launcher's does already what is requested.

lshems, 2018-09-01:
To get back to the original request. Why not have three files:

The scriptx.lua itself, a scriptx.txt containing a filename, and a scripts.png containing the image.

Only a small adaptation to the sdcard browser is neccesary, and a shortcut somewhere in the GUI.

So back to implementing the original request without lua. Seems better and simpler perhaps.

Sorry I entered Lua into the equation. As far as I'm concerned, no need for any improvement, apart from the shortcut to launch a script from a directory.

mpaperno, 2018-09-01:
> The scriptx.lua itself, a scriptx.txt containing a nice appname, and a scripts.png containing the image.

That's good too, or maybe just one ".info" metadata file which can list the name, image (maybe different sizes/versions per LCD type), and any future things we may want in the future, like author/copyright/description/etc.  If this really takes off then the same file could be used in an online repository for example (`otx install fancyapp` :-) ).

> Only a small adaptation to the sdcard browser is neccesary

Presumably we'd want to hide any of those "extra" files, but that's easy enough to do by the extension type, as long as they are consistent.

> Also,you need to think about submenus.

If you mean like "categories" or, essentially, subfolders, then yes I have... :)  On one hand it would be good and forward-thinking to have subfolders one can browse into.  OTOH if one has an app with assets (images/data/etc) then it would be best to group all those in some subfolder as well.  I can think of several ways to handle this (eg. hide folders which match a script name), but it's another thing to consider.

lshems, 2018-09-10:
https://r.tapatalk.com/shareLink?url=https%3A%2F%2Fwww%2Ercgroups%2Ecom%2Fforums%2Fshowthread%2Ephp%3Ft%3D3129894&share_tid=3129894&share_fid=8309&share_type=t

Just a link to a how-to to dump 'settings' into a file, and retrieve them with loadfile().

This is what I was referring to as advantage of using Lua interpretable settings inside the script to be read by other scripts for making this 'menuready' scripts.

mpaperno, 2018-09-10:
Yea, the meta-data file could be in Lua table format (though it would still be a separate file, as there's no way to load only some part of a larger script AFAIK).  Using Lua to parse the meta data is a nice idea... I'm not sure it would be more efficient than a special-made parser in C, but the benefits may easily outweigh that (eg. don't have to write a parser, and the data can be used directly in Lua as you said).  Maybe a bit overkill for just a name and (possible) image path, but much more flexible for future use.  And it is very easy to process on the C[++] side also, because Lua provides the API.

RealTadango, 2020-01-07:
Still relevant :)

3djc, 2020-01-07:
That’s TOOLS now isn’t it ?

RealTadango, 2020-01-07:
Ok, will check this TOOLS thing, haven't seen it :)

RealTadango, 2020-03-09:
Closed by the new Tools menu.

closed 2020-03-09 by RealTadango