opentx/opentx #4576: The X7 bars telemetry screen is not yet working

lthole, 2017-03-07:


3djc, 2017-03-07:
We will probably one if we can make a nice despite the low response, but not before 2.2.0

bsongis, 2017-03-07:
Are we talking about this screen?

[image]

3djc, 2017-03-07:
Erf misread, was thinking about chan mon

lthole, 2017-03-08:
I am unable to get that screen. I am on 2.20N363 RC11 and Win7SP1. I can get the NUMS screen, but no BARS screen.
I have just created a new model, and bound it to an X6R receiver with an SP-FLVS S.Port Lipo sensor. I ran 'Discover new sensors', and it found my sensors. I then set up a BARS screen with RxBt and Cels. From the main view, I pressed [Page_long] and got a screen saying 'No Telemetry Screens' and 'NO DATA'. I then added a NUMS screen with RxBt and Cels, and that screen comes up fine, but the NUMS still does not come up.

bsongis, 2017-03-08:
True. I will change the bug title

bsongis, 2017-03-08:
- [ ] Standard bottom bar for RSSI is too far to the right, so not displaying the value
- [ ] We have no TX RSSI, we could write "RSSI"
- [ ] Labels should be written in SMLSIZE?

3djc, 2017-03-08:
I'll take that

3djc, 2017-03-08:
[image]

With SMLSIZE:
[image]

3djc, 2017-03-08:
I prefer SMLSIZE

bsongis, 2017-03-08:
Indeed!
Would you try also the SMLSIZE for the bar label (Thr)?

3djc, 2017-03-08:
[image]

3djc, 2017-03-08:
or thi sthat I prefer:
[image]

PR #4580 (2017-03-11): X7 review telem bars screen
As per #4576 discussion and screenshots 

This fixes #4576
closed 2017-03-11 by bsongis