opentx/opentx #4582: using gVar as offset wrong scaling (x7 radio)

lshems, 2017-03-09:
When using a Gvar as offset on a mixer line, the actual effect is only 10 % of the Gvar setting.

So on a mixer line, source max, weight 0, offset gVar1 with gVar set to 100, the output is 10.

lshems, 2017-03-09:
Sorry, only when using -gVar as offset this is not working correct.

gVar as offset with a negative value of gVar is working correct.

So must be in the - part of the gVar thing.

lshems, 2017-03-10:
It was late yesterday. Correct issue:

Gvar 100

Source Max, weight 0, offset -gvar
Result -10 instead if -100

Source Max, weight 0, offset gvar
Result 100, as expected.

3djc, 2017-03-10:
[image]

[image]

I don't seem to be able to replicate, could you let me have your otx ?

lshems, 2017-03-10:
Ah, you'r on RC11, I had returned to RC10, because the RC11 had some other issues bothering me. Forgot to test on RC11 as well.

So, it seems that it is fixed on RC11. 

Thanks.

lshems, 2017-03-10:
Could we reopen: tested on RC11, and got a similar result.

test otx attached.

just change the offset from -gv1 to gv1 and see the difference
[image]
[image]

[testerror.zip](https://github.com/opentx/opentx/files/834958/testerror.zip)

lshems, 2017-03-10:
tested on taranis 2.2RC11, same issue. I have no clue what happens, since a new model with the same settings doesn't exhibit this behaviour; Only the attached OTX does this.

kilrah, 2017-03-10:
Weirdo

lshems, 2017-03-10:
Indeed.

lshems, 2017-03-10:
remark that the gvar screen is showing only zero's !?

kilrah, 2017-03-10:
Hmm not here?

lshems, 2017-03-10:
[image]

lshems, 2017-03-10:
[image]

lshems, 2017-03-10:
perhaps that screen is not functional yet, I don't know.

kilrah, 2017-03-10:
Indeed, didn't see it...

Open companion, open file, click simulate (on document) - values are shown correctly (but output is bad). Close simu, click simulate again, the<y're all 0.

@mpaperno, help? haven't followed storage evoltion close enough, but seems like it could be related?

mpaperno, 2017-03-11:
Indeed, the Simulator Radio Outputs gvars not updating was a bug (fixed @ 0a5732b953e1a87763a6ae06ea946c843f6ecbf1).  Thanks for noticing  :)

No clue on the problem of the actual gvars calculations in firmware...  I can't keep track of the import/conversion routines either, if that's what you mean Andre.  I guess first step is consistent repro case.  Maybe @bsongis has some clues for us.

kilrah, 2017-03-11:
No I just meant the bug you fixed :)

lshems, 2017-03-14:
Ok, found a reproducable action triggering the issue. It doesn't make sense at all:

in the attached file, there are two models, both having input1 with SA as source, gVAR1 having a value of 50, and mixer line 1 having source Max, weight 0 and offset GVAR1. mixer line 2 having source Max, weight 0 and offset -GVAR1.

When no switch warnings are active, the second mixer line produces -5, where it should have produced -50.
When activating switch warning for SB, the second mixer line is correctly showing -50.

Tell me whats happening ???

[testerror (2).zip](https://github.com/opentx/opentx/files/842922/testerror.2.zip)
[image]
[image]

3djc, 2017-03-15:
I can reproduce that, will be looking at that today

lshems, 2017-03-15:
Thanks. It is a plane crashing one for my models after conversion from X9, if I wouldn't test properly before using it :).

PR #4623 (2017-03-15): Handle neg gvars precision
This fixes #4582
commit 8a84c215 (2017-03-15): Fix #4582 (#4623)
closed 2017-03-15 by bsongis