opentx/opentx #4583: Companion new document could use the selected profile data

kilrah, 2017-03-10:
Automatically load radio settings from profile when creating new document

mogplus8, 2018-06-03:
Not sure if my problem is the same thing, but I have found a problem with profiles.
I have two radios, a Taranis and a Turnigy 9XR Pro, both running OpenTX 2.2.1.
I have found that if I forget to choose the correct radio profile from the button on the toolbar or from Settings / Radio Profiles and the wrong one is still selected from the last session, when I open the .otx file, Companions offers to convert my models to the profile format selected in the dropdown, and does *not* give me the option not to. Wouldn't it be easier if it just used the profile specified in the .otx, and, if required, allow the profile to be changed by selecting a different one in the radio setup where it is already specified? That way the selection of the profile would not be needed at all, just the functionality to create, update and delete (unused) profiles. And no more selecting the wrong one and maybe accidentally converting all your planes to the wrong format.
Thanks,
Ian

kilrah, 2018-06-03:
Profiles are not stored in data files so cannot automatically loaded, only the radio type is known but since you can have 3 different profiles for the same radio type companion can't know which one you want to load.

When you get the warning you can jsut close, sleect the right one and reopen. Nothing is lost unless you saved inbetween...

And no, it's not related to this issue at all.

mpaperno, 2018-06-03:
Proposal: We do away with the whole "store setting in profile" bit and instead let the user specify a default template file to use when creating a new file.  One template per radio profile.

Reasoning: 
 1) The current scheme only stores some hardware settings and calibration, not the other zillion options.
 2) Maintaining the copy in profile is extra work for both devs and users (I'm not sure it even works very well right now, and the code is ancient).  User can't even see what they have stored.
 3) The template could have not only default radio settings but also model(s).

mpaperno, 2018-06-03:
@mogplus8 The process is much improved in v2.2.2. You now get some options when opening a file which doesn't match the radio in the current profile (or when switching profiles), instead of automatic conversion. It can be improved further, but ultimately we can't tell exactly which profile the file was saved from, as @kilrah said.

mpjf01, 2018-06-03:
Thank you. The option not to convert is an improvement.

mogplus8, 2018-06-04:
Thanks for the responses. I can't comment (or understand) the technical issues or implementation, nor am I familiar with all the use cases. I can only provide my experience as an end user.
I think (from my limited understanding) that @mpaperno may be on the right track. Maybe it might be preferable, from a user perspective, to manage all the radio settings in the Radio Settings function, including multiple radio profiles, rather than having them stored and managed separately.
And apologies for the original post in the wrong place. 
Thanks,
Ian

mpaperno, 2018-06-05:
Regarding removing the stored settings in profiles, I also see they're used (or were used) in the "eeprom" flashing dialog to "patch" the data before flashing.  I don't really understand why you'd want to do that (I'm guessing a legacy feature?), and besides the code is again very old and I doubt it even still works in a lot of cases (I couldn't get the options to become enabled at all, and looking at [the code](https://github.com/opentx/opentx/blob/2.2/companion/src/flasheepromdialog.cpp#L82) I can see why).

kilrah, 2018-06-05:
Radio A, with profile and stored settings.
Radio A's profile is selected.

Read radio B. Document contains radio B's settings. Click write with the patch checkbox enabled, the document from radio B is writtin to radio A but with the hardware-related settings of radio A.

Quick way to make a "perfect" copy of a radio to another without touching what is hardware related and may be different.

But yeah it's been unmaintained and would need checking the choice of settings is still relevant...

mpaperno, 2018-06-05:
What happens when radio B has different hardware config than what the models expect?

And can't that be done by copy/pasting the radio settings between files?  Or maybe I'm still missing the point.

Unrelated, but makes me think it would be nice to have a "verify settings" feature which works a bit like the conversion routine -- goes through and verifies that all the hardware/etc defined in the models still exists in the radio config (eg. after copy/pasting settings from another radio).  And for that matter that all the inputs/mixes/LS/etc/etc are still valid.

mpaperno, 2018-06-06:
I guess this issue is technically a bug... because the code is already there but it just doesn't work (same issue as in all the other places... unmaintained).  :)  
https://github.com/opentx/opentx/blob/2.2/companion/src/firmwares/generalsettings.cpp#L159

That's what... 3 places the non-working code is now duplicated?  Refer to item 2) on my list above  :)

mpaperno, 2018-06-06:
Related: #1705

kilrah, 2018-06-06:
> What happens when radio B has different hardware config than what the models expect?

Well there wasn't anything that could be problematic back then... But yes now the switch/pot settings could be an issue

closed 2020-04-06 by stale[bot]