opentx/opentx #4585: ENTER/MOUSE-MID defined in the the simulator help is not correct

LapinFou, 2017-03-10:
When doing the French translation (and testing the RC11 at the same time), I discovered that the help is wrong.
**ENTER/MOUSE-MID** should be **ENTER/MOUSE-LEFT**

mpaperno, 2017-03-11:
Mouse left button?  That doesn't change normal meaning.  Mouse MID will (or should) activate the radio "action" (ENT/MENU/Rotary click) while anywhere over the Simulator window.  So I'm not sure what you mean.

That said, I did find some issues in the help text and the controls themselves, which I have changed already in my working copy. I'm also centralizing all the strings in one file to make it simpler to maintain/translate.  So don't spend much time translating the keyboard help text right now.

E.g. from X9E:
[image]

What do you think of the icons?

kilrah, 2017-03-11:
> ENTER/MOUSE-MID should be ENTER/MOUSE-LEFT

Umm no, it's correct... :\

> What do you think of the icons?

Yep why not!

LapinFou, 2017-03-11:
Well I guess I didn't understand what this **ENTER/MOUSE-MID** is for.
With the X9D+ simulator, the middle click doesn't work on the 6 buttons (PAGE, MENU, etc...). But, the left click does.
However with the LUA script (for example the Wizard), I can scroll with the mouse to select, then use the middle button to confirm the action.

kilrah, 2017-03-11:
Left-click on the buttons activate them, including the rotary encoder "enter" push.
Mid mouse activates the rotary encoder push regardless of where the cursor is.

mpaperno, 2017-03-12:
Ah, I see... ironically, mid-clicking anywhere BUT in the button areas works correctly.  But while cursor is over the control areas on the left/right, mid-click doesn't actually work at all.  I think I also know why... yes, much better now (fixed... porting... testing... committing...).

closed 2017-03-12 by mpaperno

mpaperno, 2017-03-12:
Now if mid-clicking on a radio key, that key is activated.  Otherwise default mid-click action is triggered (enter/rotary click/etc).  I tried it so the default action is always triggered, but it was awkward and counter-intuitive when mid-clicking on a different key and instead "enter" was being pressed.

commit f103c885 (2017-03-12): [Simulator] Improve middle mouse button handling over radio navigation key areas: clicks were ignored, now either activates the current key under mouse, if any, or triggers default middle-mouse action (enter/menu/rotary press). (closes #4585)