opentx/opentx #4588: Companion build error (on bracnh NEXT)

merinkumar, 2017-03-11:
Hi All, Since there are no 64 bit rpm version of open tx companion, i tried building using https://github.com/opentx/opentx/wiki/Companion-Linux-Build-Instructions a 64 bit version. Build fails.. when i reach this step below... (How ever when i tried master branch it works perfectly, but when i do that, companion dont show TARANIS QX7 in menu to select my radio target). Please help

ERROR>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
lbuildmerin> cmake ../src
-- Simulators library search path: /usr/local/lib/companion
CMake Error at CMakeLists.txt:98 (use_cxx11):
  Unknown CMake command "use_cxx11".

-- Configuring incomplete, errors occurred!
See also "/home/merin/opentx/companion/lbuild/CMakeFiles/CMakeOutput.log".
See also "/home/merin/opentx/companion/lbuild/CMakeFiles/CMakeError.log".

3djc, 2017-03-11:
On the page you link, there  nice link on the right side called something like 2.2 Linux compile...

closed 2017-03-11 by 3djc

mpaperno, 2017-03-11:
For `next` branch use these instructions: https://github.com/opentx/opentx/wiki/OpenTX-2.2-Linux-Build-Instructions

All the official OTx Linux builds are 64-bit.

merinkumar, 2017-03-12:
thank you @3djc , @mpaperno  actually I saw that and tried that too now am struck at arm corss compiler, as i dont have it installed, and am trying figure how to install that.

@mpaperno you mentioned "All the official OTx Linux builds are 64-bit." but when i checked in downloads, for ubuntu there was 32 and 64 but for rpm (centos) only i686 , Please let me know if am looking at different place...

mpaperno, 2017-03-12:
Sorry, I meant the 2.2 beta releases (though one can get a 32-bit version via PPA). There are no rpm versions that I can see.  I guess it's "Debuntu" only.  
http://www.open-tx.org/2017/03/05/opentx-2.2.0RC11  
http://downloads-22.open-tx.org/nightly/companion/linux/

adelton, 2017-08-20:
I have built 2.1 and 2.2 version rpms of Companion at https://copr.fedorainfracloud.org/coprs/adelton/opentx-companion/ and https://copr.fedorainfracloud.org/coprs/adelton/opentx-companion-2.2/.

The .spec's and .src.rpm's with the patches are included in the build result so it should be easy to see how the build was done.