opentx/opentx #4589: Horus crashing on model image selection

joaquins, 2017-03-12:
Hi,

Just upgraded my Horus to firmware:
opentx-x12s-ppmus-multimodule-noheli-lua-massstorage-en-2.2.0N363.bin

When selecting a new image model, the radio crashed under these conditions:
- Case one: Quick scroll backwards of model images, radio shows a red block covering the lower end of the screen, and becomes non-responsive. Needed to shutdown with long-press on power button.
- Case two: Quick scroll forward of model images. Radio reboots by itself into "emergency mode". Needed to shutdown with long-press on power button

- Case three: Slow scroll of image models, after scrolling last image, radio stop responding, and shutdown is needed.

Attaching my images folder in case it helps.
[IMAGES.zip](https://github.com/opentx/opentx/files/836215/IMAGES.zip)

kilrah, 2017-03-12:
Scrolling where? On the model setup page or in the SD card browser?

Can't reproduce on either even with your folder. Coudl be bad SD card...

marcogpw, 2017-03-12:
I've had the same happen to me: on the Horus trying to change the model picture via the telemetry menu, it crashed on me as well, showing the emergency mode screen

3djc, 2017-03-12:
Changing model picture using telemetry menu ????

marcogpw, 2017-03-12:
long press telemetry on the right selector, then you get the screen where you can change the widgets for telemetry etc, one of the screens holds the model picture selected in the main model setup and you can change it there as well, during the picture change it froze on my Horus

3djc, 2017-03-12:
hum, how exactly to you change the picture there ?? It is a display only as fas as I know

joaquins, 2017-03-14:
Update: I found some hidden files under my IMAGES folder. Files starting with "._", created by OSX.

After deleting those files, I can't reproduce the crash any longer.

bsongis, 2017-03-14:
are you able to reproduce this issue? would you give us an example of a file which does this?

marcogpw, 2017-03-14:
OK, update on my crash event: I checked the IMAGES folder, there are no strange files in there. I tried some more things and I found the culprit: the telenowa script. As soon as I am setting widgets and the telenowa widget was in the selection box the Horus crashed and Emergency mode was on the display. Most probably this is what happened last time as well. Deleted this widget, so I expect no more issues. Thanks for all your thinking guys !

projectkk2glider, 2017-03-14:
@marcogpw the telenowa script issue should be handled in RC12, which version are you using?

marcogpw, 2017-03-14:
RC12 is on my Horus since yesterday, tried just 5 minutes ago, is crashed twice while the telenowa widget came by, scrolling trough all widgets, it wasn't even selected yet, just showing as an option while scrolling was enough for the crash. I downloaded the telnowa widget a while ago, is there a chance that this was an old version ?

projectkk2glider, 2017-03-14:
Interesting, every model load uses some RAM, this is used heap size, ever line is after a new model load:
 ```
	used  1455104 bytes
	used  1864704 bytes
	used  2028544 bytes
	used  2192384 bytes
```

closed 2017-03-22 by projectkk2glider

br34k0r, 2019-09-10:
This issue appears again with the X10S Express. Also when selecting a .wav file from play sound.
(Latest Nightly)

When you set it via companion, it works.

commit 618345a2 (2017-03-22): Fixes #4589 (#4644)

Produces a segfault with the telenowa script (missing bitmap)