opentx/opentx #4596: Again Picasso effect in Model folder with model pictures Horus 2.2.363

Githubbachelor, 2017-03-13:
Again Picasso effect in Model folder with model pictures Horus 2.2.363.

If you are choosing / opening / loading about 15 models after and after, come a moment where the pictures from models in the folder to choose a model goes like picasso in all colors points. Only if want to control all your models and you open them after and after.

kilrah, 2017-03-13:
Are you using any third-party scripts / widgets in some of these models? Problem has been found to be casued by a buggy widget that eats up all memory and should not be used.

projectkk2glider, 2017-03-13:
The script problem should be fixed by #4284. I think we got some other problem here. There was also a report that browsing trough model images also crashes the radio - emergency mode..

projectkk2glider, 2017-03-13:
@Githubbachelor is the result the same as the photos you posted in #4072?

That corrupted images look all the same - which is a bit unusual.

projectkk2glider, 2017-03-13:
In fact we still have open issue for this #4099, closing as duplicate.

closed 2017-03-13 by projectkk2glider

Githubbachelor, 2017-03-13:
Yes it is the same result as #4072 and Ihave no other widgets