opentx/opentx #4598: ..companion22_2.2.0N364_amd64.deb released as "companion22_2.2.0.deb", cracked?

gasguru, 2017-03-13:
..companion22_2.2.0N364_amd64.deb was released today as "companion22_2.2.0.deb", failed build or cracked build system?

.."companion22_2.2.0.deb" posted on http://downloads.open-tx.org/2.2/nightly/companion/linux/, but is supposed to be "companion22_2.2.0N364_amd64.deb" mentioned on http://www.open-tx.org/2017/03/13/opentx-2.2.0RC12, INFO file in "companion22_2.2.0.deb" mentions "Version: 2.2.0N364", so no joy quite yet.

..correct language in the names of .deb files is $packagename_$version_$arch.deb, here "companion22_2.2.0.deb" lost its version number "N364", the under score "_" and the architecture pointer "amd64", and may need a full rebuild if my simple mv fails, I did:
$ ll -n companion22_2.2.0.deb &&md5sum companion22_2.2.0.deb
-rw-r--r-- 1 1000 1000 14634620 Mar 13 10:54 companion22_2.2.0.deb
79d6590f0619812279fe04a1b5a027ff companion22_2.2.0.deb

$ mv -vf companion22_2.2.0.deb companion22_2.2.0N364_amd64.deb 
'companion22_2.2.0.deb' -> 'companion22_2.2.0N364_amd64.deb'

$ ll -n *.0N364* &&md5sum *.0N364*
-rw-r--r-- 1 1000 1000 14634620 Mar 13 10:54 companion22_2.2.0N364_amd64.deb
79d6590f0619812279fe04a1b5a027ff companion22_2.2.0N364_amd64.deb

gasguru, 2017-03-13:
..installs fine:
root@celsius:/var/cache/apt/archives# dpkg -BEGi\
 companion22_2.2.0N364_amd64.deb 
(Reading database ... 2912369 files and directories currently
installed.) 
Preparing to unpack .../companion22_2.2.0N364_amd64.deb ...
Unpacking companion22 (2.2.0N364) over (2.2.0N358) ... 
Setting up companion22 (2.2.0N364) ... 
Processing triggers for hicolor-icon-theme (0.15-1) ... 
Processing triggers for gnome-menus (3.13.3-9) ... 
Processing triggers for desktop-file-utils (0.23-1) ...
Processing triggers for mime-support (3.60) ...
root@celsius:/var/cache/apt/archives# 

..appears to work ok, started up using Taranis X9D+ rather than my X9D:
:~$ [D] registerStorageFactory():54 - register storage "bin"
[D] registerStorageFactory():54 - register storage "eepe"
[D] registerStorageFactory():54 - register storage "hex"
[D] registerStorageFactory():54 - register storage "otx"
[D] registerStorageFactory():54 - register storage "sdcard"
[D] registerSimulators():63 - Searching for simulators in "."
[D] registerSimulators():63 - Searching for simulators in "/usr/local/lib/companion22"
[D] registerSimulator():39 - trying to register simulator in  "/usr/local/lib/companion22/libopentx-9x-simulator.so"
[D] registerSimulator():44 - Registered "opentx-9x" simulator
[D] registerSimulator():39 - trying to register simulator in  "/usr/local/lib/companion22/libopentx-9xrpro-simulator.so"
[D] registerSimulator():44 - Registered "opentx-9xrpro" simulator
[D] registerSimulator():39 - trying to register simulator in  "/usr/local/lib/companion22/libopentx-gruvin9x-simulator.so"
[D] registerSimulator():44 - Registered "opentx-gruvin9x" simulator
[D] registerSimulator():39 - trying to register simulator in  "/usr/local/lib/companion22/libopentx-mega2560-simulator.so"
[D] registerSimulator():44 - Registered "opentx-mega2560" simulator
[D] registerSimulator():39 - trying to register simulator in  "/usr/local/lib/companion22/libopentx-sky9x-simulator.so"
[D] registerSimulator():44 - Registered "opentx-sky9x" simulator
[D] registerSimulator():39 - trying to register simulator in  "/usr/local/lib/companion22/libopentx-x10-simulator.so"
register theme Darkblue
register theme Default
register layout Layout1x1
register layout Layout2P1
register layout Layout2x4
register layout Layout2x2
register layout Layout2x1
register widget Outputs
register widget Value
register widget ModelBmp
register widget Text
register widget Gauge
register widget Timer
[D] registerSimulator():44 - Registered "opentx-x10" simulator
[D] registerSimulator():39 - trying to register simulator in  "/usr/local/lib/companion22/libopentx-x12s-simulator.so"
register theme Darkblue
register theme Default
register layout Layout1x1
register layout Layout2P1
register layout Layout2x4
register layout Layout2x2
register layout Layout2x1
register widget Outputs
register widget Value
register widget ModelBmp
register widget Text
register widget Gauge
register widget Timer
[D] registerSimulator():44 - Registered "opentx-x12s" simulator
[D] registerSimulator():39 - trying to register simulator in  "/usr/local/lib/companion22/libopentx-x7-simulator.so"
[D] registerSimulator():44 - Registered "opentx-x7" simulator
[D] registerSimulator():39 - trying to register simulator in  "/usr/local/lib/companion22/libopentx-x9d+-simulator.so"
[D] registerSimulator():44 - Registered "opentx-x9d+" simulator
[D] registerSimulator():39 - trying to register simulator in  "/usr/local/lib/companion22/libopentx-x9d-simulator.so"
[D] registerSimulator():44 - Registered "opentx-x9d" simulator
[D] registerSimulator():39 - trying to register simulator in  "/usr/local/lib/companion22/libopentx-x9e-simulator.so"
[D] registerSimulator():44 - Registered "opentx-x9e" simulator
[D] MainWindow::checkForUpdates():218 - Checking for Companion update  "http://downloads-22.open-tx.org/companion/companion-windows.stamp"
[W] unknown():0 - QFSFileEngine::open: No file name specified
[D] main():143 - COMPANION EXIT 0

[1]+  Done                    companion22

3djc, 2017-03-13:
Was just  renaming issue in the scripts

closed 2017-03-13 by 3djc