opentx/opentx #4599: [Companion 2.2.RC12] no separation between the mixes lines

LapinFou, 2017-03-13:
Since the release 2.2.RC11 (Win10 env), there is no more separation between mixes lines.
I don't know if it's a small cosmetic bug, or if it's done in purpose. However, it was better before 😉 
Moreover it is inconsistent with the **Inputs** tab.

See the picture
[image]

projectkk2glider, 2017-03-13:
Just for additional info, different QT version will produce different result, here is QT 5.2.1 on Linux.

Here the separation is OK, but lines are truncated at the bottom!
[image]

The QT 5.7 (non-native for me) produces this:
[image]

mpaperno, 2017-04-30:
I've looked briefly at this a couple times but so far can't even understand why the fonts are different on Inputs vs. Mixes screens.  Some day....

LapinFou, 2017-04-30:
The font doesn't look much different between OTX 2.1 and OTX 2.2.
I thought the only difference was a different interline parameter (or similar).

See on the screenshot below, on the top there is a EOL (or similar) between each mixer line, but not on the bottom screenshot.

[image]

mpaperno, 2017-04-30:
Sure, the document formatting is "quite" different.  The font difference is either the reason or a symptom, not sure which.  The 2.1 version of Mixes looks like the same as 2.2 Inputs format.

PR #4919 (2017-06-10): [Companion] Inputs and Mixes lists formatting & DnD fixes:
  * Fix/improve margin/padding issues, consistent between inp/mix views, better handling of x-platform and older Qt (closes #4599, closes #4341);
  * Allow dropping onto lines instead of only between them;
  * Fix dropped data always going to line above the actual drop target (closes #4622);
  * Force DnD _move_ operations between windows to _copy_ data instead (fixes deletion of any selected line in target window, and source line was never actually removed but looked like it was);
  * Add alternate-row coloring for each Input/Mixer gr [... cut]
commit 3f5a3995 (2017-06-10): [Companion] Inputs and Mixes lists formatting & DnD fixes: (#4919)

* Fix/improve margin/padding issues, consistent between inp/mix views, better x-platform and older Qt (closes #4599, closes #4341);
    * Allow dropping onto lines instead of only between them;
    * Fix dropped data always going to line above the actual drop target (closes #4622);
    * Always force DnD move operations between windows to copy instead (fixes deletion of any selected line in target window, and source line was never actually removed but looked like it was);
    * Add alternate-row coloring for each Input/Mixer g [... cut]
closed 2017-06-10 by bsongis