opentx/opentx #4600: Trim active in the Simulator even if their are disabled

LapinFou, 2017-03-13:
When simulating my X9D+ (OTX 2.2.RC12) I saw than I can move the trims (Ail, Ele, Thr, Rud) in the simulator even if they are disable in the FM.
With OTX2.1.9, the trim cursor is always going back to neutral position if its associated trim is disabled.

mpaperno, 2017-03-22:
Yes, the old version would continually update the trim sliders (even if they didn't change), but this was optimized out in favor of using less CPU cycles (checking first if there were any changes).  This issue does point out a drawback in that paradigm though,  But I think I have a solution which will be possible in the next version.

LapinFou, 2017-03-22:
Thanks for your time! 👍 
Can wait to see the first official 2.2.0 version. Companion has been so much improved. You rock! 😄

LapinFou, 2017-03-23:
I checked the RC14 (N366). When clicking on the left/right up/down trims in the simulator, if the trims are disabled, they don't moved anymore. However if you move the corresponding slider with the mouse, it will not go back to the middle position. 
So it is partially fixed.

mpaperno, 2017-03-23:
Oh I didn't even realize the trim buttons were also a problem before (when I tested they were properly disabled so I assumed the problem was just the slider).  Good to know, thanks for the update (and the flowers :) )!

LapinFou, 2017-03-23:
You're very welcome! 😃

closed 2017-04-02 by bsongis