opentx/opentx #4601: GitBook 2.2 List of valid sources

RCdiy, 2017-03-13:
As per projectkk2glider on rocket chat

Looking at getValue() documentation in 2.2 Gitbook. Is there an updated list of valid sources for 2.2?

https://opentx.gitbooks.io/opentx-2-2-lua-reference-guide/content/general/getValue.html

3djc, 2017-03-13:
Since you can choose your sensors name, and now also create your own, it's a kinda tough to have a list

projectkk2glider, 2017-03-13:
@3djc his request is valid, the list is listing fixed known sources for each radio, for example for X9E http://downloads-21.open-tx.org/firmware/lua_fields_taranis_x9e.txt

3djc, 2017-03-13:
Sure,  it just need to be clear that it is not a full list of valid sources. I like your wording tho (listing of fixed known sources)

RCdiy, 2017-03-13:
I was trying to figure out how to get to the tx battery voltage and found it on the list, very last line:)
So my immediat need was met bit figured for consistency a 2.2 list would be good even if it is identical to the existing lists:)

3djc, 2017-03-21:
fixed wth https://github.com/opentx/opentx/commit/1f1098e270d69068fdc9ce692e86c4f0bfb60ccf

closed 2017-03-21 by 3djc