opentx/opentx #4604: QX7 - display dark after update to 2.2.0 RC12 (N364)

franz4949, 2017-03-13:
After update to nightly 2..2.0 RC12 (N364) and pressing "Exit" the display remains dark and also switch off button does not work, battery must be disconnected and reconnected. After that display remains dark but Tx starts up and can then also be switched off via button hold pressed. Start in boot mode does work normal and can be connected to companion. I read the Tx settings with companion, changes values of contrast and brightness, wrote settings back into Tx, however no success. After "Exit" of bootmenu display remains dark and power button again without function. Only way left is disconnect/reconnect battery.

3djc, 2017-03-13:
What is the name of the firmware you flashed ?

franz4949, 2017-03-13:
Version is as written in heading, thus nightly build 2.2.0 RC12 (N264).
And Tx display remains dark when started in normal operation mode, however possible to shut off by pressing and holding power button. Display only illuminates normal when started in bootloader mode.

3djc, 2017-03-13:
What we need is something like opentx-x7-lua-massstorage-en.bin

franz4949, 2017-03-13:
Ok => opentx-x7-noheli-lua-massstorage-de-2.2.0N364.bin

3djc, 2017-03-13:
That's interesting, since that specific firmware has not been compiled on our server yet

franz4949, 2017-03-13:
I took the windows installer (companion) from here => http://www.open-tx.org/2017/03/13/opentx-2.2.0RC12#

3djc, 2017-03-13:
OK, I found it, sorry. All seems fine on our side, I'll try flashing it on my radio

3djc, 2017-03-13:
Works fine here (just a strange language :D)
Have you tried working your way to lcd settings just in case ?

franz4949, 2017-03-13:
Sorry for the strange language :D 
Only way to do this is start in bootloader mode, connect to companion, read models and setup from Tx (Q7), do some setup changes for contrast and brightness and write thenchanges back into Tx. This way worked for the previous nightly build but does not work with actual version. Also i should mention that none of the two display criteria were down with its values. Brightness was/is at 100%, contrast at 35%.

projectkk2glider, 2017-03-13:
@franz4949 is your radio really hot or cold? Try starting without the SD card. Flash the radio again.

franz4949, 2017-03-13:
Have taken out SD card and then flashed again, no change in result. After leaving bootloader mode, display remains dark, no function of switch on/off power button. After dis/reconnection of battery I can start Tx but display remains dark. Power button works to switch off and I can start bootloader mode normally.

franz4949, 2017-03-13:
Radio has normal temperature, thus not hot not cold.

projectkk2glider, 2017-03-13:
Can you flash back the RC11 just to confirm that it works - ie that your radio has not suffered a hardware problem.

franz4949, 2017-03-13:
I flashed 362 (must be RC10 I think) and it works normal. With RC11 I I had same problem as with RC12, only difference was, that after reading of Tx setup via companion and writing back again the display worked. With RC12 it does not work this way.

projectkk2glider, 2017-03-13:
This would suggest that the issues is indeed the contrast setting.

franz4949, 2017-03-13:
But what about the frozen status after leaving the bootloader mode? Power button without function, only thing I can do is disconnect battery.

projectkk2glider, 2017-03-13:
Previously you wrote:

>And Tx display remains dark when started in normal operation mode, however possible to shut off by pressing and holding power button. 

Do you hear any trim beeps when your use trims?

franz4949, 2017-03-13:
And if the contrast is down, should I not see this then when I read the Tx setup with companion?

When I start Tx normal, yes then display is dark but power button does work, means I can switch Tx off. However when Exit bootloader mode Tx is frozen and does not start,

projectkk2glider, 2017-03-13:
Try setting the radio language to English (in Companion), then download the new firmware and flash the radio.

franz4949, 2017-03-13:
I did but no change in result.

franz4949, 2017-03-13:
Again after Exit bootloader mode only way is disconnect and reconnect battery. After then normal start display dark, no sound when switching trims, however can switch off with power button.

franz4949, 2017-03-13:
What can I do to give you more data/information?

projectkk2glider, 2017-03-13:
I would like to have a confirmation of the same issue from at least one another user. So far it looks like your specific radio problem.

franz4949, 2017-03-13:
I understand what you say, but I do not understand how this might be specific for just for my radio. The radio I received from Banggood some 3 weeks ago. I am not using it for flying yet, I also have two X9D radios. I only flashed german OTx nightly RC10 which worked normal. Then some days ago I flashed RC11 and run first time into this problem, which I could resolve by Companion reading Tx setup and writing it back again. And now with RC12 also this does not work. That is all what I did up to date with this radio, no use for flying no mods. nothing.

franz4949, 2017-03-13:
Strange. Anyway I'd like to thank you for your efforts to resolve this matter.

franz4949, 2017-03-13:
Additional information: Bootloader on my QX7 is =>  X7   -  2.2.0N362

3djc, 2017-03-14:
While it sounds logical, you are the only one experiencing it, which in return points to a problem on your side.  Could you try the following : put RC (or whatever works), make 100% sure you have backups of your radio, long press menu, go to page 5/9, enter long for a factory reset, then try to flash RC12.

franz4949, 2017-03-14:
Problem solved by installing the update **not via companion** but **direct from SD card**. This proposal I got from a guy in RCG who reported that he was facing the same problem as I had and he solved it in the proposed way by direct update from SD card.

So for me I am fine and for me I can close this issue. If there is something not so perfect with companion, I cannot say, maybe you might have a closer look there.

Many thanks again for your prompt support and effort to help me.

3djc, 2017-03-14:
Happy to see it solved  !

closed 2017-03-14 by 3djc