opentx/opentx #4606: Crossfire protocol on external turn off internal Module

Webbsolution, 2017-03-13:
I understand that previous version of the Crossfire firmware left the telemetry hard wired "on" which would lead to a data conflict when its telemetry shared the same line as the Internal Frsky Telemetry. As I understand it, that is why the internal module turns off when the external module is set to Xfire - Now that the current Xfire firmware revs allow the user to disable telemetry, would it be possible to use both modules at the same time WHILE enabling the Xfire protocol ?

schwabe, 2017-03-13:
The communication to the module still runs via the sport telemetry pin which is shared with the internal module, so in short no.

closed 2017-03-13 by schwabe

Webbsolution, 2017-03-13:
then how is it possible to run the Crossfire in PPM mode without the same problem ?

schwabe, 2017-03-13:
that uses the ppm pin

Webbsolution, 2017-03-13:
so -- if the crossfire protocol was using that pin ? could we have a solution ?

schwabe, 2017-03-13:
Actually older versions of Crossfire (as implemented in OpenTX 2.1) used the PPM as input. But with the support of Telemetry/OpenTX 2.2 it changed to bidirectional communication over the s.port pin.

I am not involved with TBS or the OpenTX Crossfire, so I have no idea if that is going to change back or if a second crossfire protocol for ppm will be implemented (but I would guess not).

Webbsolution, 2017-03-13:
thank you for the clarification. PPM works and I guess thats all I can expect. If I want the faster protocol I guess ill have to drop the Frsky internal and redundancy (using the RB10 atm) and add a S.bus to PWM converter.. ugly.

schwabe, 2017-03-13:
what happens on the rx side should not be affect by the protocol on the TX side.