opentx/opentx #4612: Curves not displayed in the "print" output

LapinFou, 2017-03-14:
Hi Folks,

I'm running the N364 on a Win10 64bits. When I do print a model, the curves are not displayed.
Here is what I got, instead the graphical views:
[image]

projectkk2glider, 2017-03-14:
Same problem on Linux.

mpaperno, 2017-03-14:
Yep, saw that while making the model printer (and the rest of Companion) show user-configured control names.  Wasn't sure if it ever worked, but I guess so?   Looks like the image isn't being properly embedded in the resulting HTML doc...  will check into it more.

LapinFou, 2017-03-14:
Never checked on previous Companion 2.2 version, but it is working on Companion 2.1.9

mpaperno, 2017-03-14:
Maybe it was a Qt 4 -> 5 migration glitch.  Confirmed working now in all 3 OSs.  Though I still need to check Damjan's favorite Qt 5.2...   😛

closed 2017-03-14 by mpaperno

mpaperno, 2017-03-14:
Yes, even there  :)

[image]

LapinFou, 2017-03-14:
That was a very fast fix!
Thanks 👍

mpaperno, 2017-03-14:
We aim to please.  Especially translators.   :-)

LapinFou, 2017-03-15:
Thanks! ^_^

commit 2fb92e3d (2017-03-14): [Companion] Fix missing curve image(s) in model printer (fixes #4612).