opentx/opentx #4613: Option FrSky D (Cable) missing in HORUS X12S, X9D, OpenTX 2.2.0N363 and companion

josch1488, 2017-03-14:
Just found a issue In OpenTX 2.2.0N363 for HORUS X12S, X9D and companion 2.2
On the Telemetry screen HORUS X12S and companion the option FrSky D **(Cable)** is missing !

X9D+ and X9E are working properly.

This option is necessary for HORUS X12S and the M-Link/FrSky telemetry converter to get the telemetry data into the HORUS.

kilrah, 2017-03-14:
Frsky D (Cable) is an option that only applies to the 9XR-Pro.

Hartinger, 2017-03-14:
It would be a shame to loose this option, as it is the basis for a number of DIY telemetry converters, including my own M-Link Converter.
The same issue was raised against 2.1.8 (#3372), which was acknowledged as bug and fixed in 2.1.9.
(The difference being that the previous issue only affected Companion but not the radio.)
Why is this no longer considered as a useful feature?
It would be very regrettable if this feature now disappears.

schwabe, 2017-03-14:
X9D+ still has the option if the 2nd serial port under general settings is configured to telemetry.
[image]

For Horus I think the mainboard has an upopulated 2nd serial header but I think the feature has not yet been implemented. (And has definitively lower priority as it is not a simply accessable as on the X9D)

nostromo123, 2017-03-14:
@kilrah.
Please rethink your statement here. I am one of the happy telemetry users with MPX receivers, oXs sensors and a M-Link converter with my X9D under openTX 2.1.8
Might have a look here **"companion 2.1.7 missing telemetry protocol choice #3372"**

schwabe, 2017-03-14:
@nostromo123 the feature is still there and selectable on X9D. For Horus see my comment.

kilrah, 2017-03-14:
Sorry got confused, I thought that in the case of the Taranis' serial port it worked by default as long as Hardware was set tot Telemetry, without needing to set the FrSky D (cable) option on the telemetry page, but you indeed have to.

But there still is no issue since you can still select it when applicable (serial port set to telemetry, internal module off, external module set to PPM).

For Horus as @schwabe said it is not supported in the firmware.

Hartinger, 2017-03-14:
> Why is the module bay pin not appropriate?

Because the M-Link module occupies the module bay.

I have no problem because for both X9D+ and X9E the feature is still there.
(And anyhow I have no intention to change to 2.2 in the near future.)
I can understand that it is low priority for the Horus.
But why is the selection of FrSky D (cable) not available for the X9D (without +) in 2.2.

kilrah, 2017-03-14:
> But why is the selection of FrSky D (cable) not available for the X9D (without +) in 2.2.

It is there...

> Because the M-Link module occupies the module bay.

Even more so, you could make a clean setup without external cabling...

Hartinger, 2017-03-14:
OK, I have just tried again with Radio type set to X9D (no plus).
The feature is indeed available, contrary to my statement above, sorry for the confusion.
I guess I had both the Radio Settings and Model Settings window open when I tried first.
If I close the Radio Settings window, after having set Serial to Telemetry, and then open the Model Settings, the option is available.

closed 2017-03-22 by kilrah