opentx/opentx #4614: Companion build from source on fedora -  No cmake_minimum_required command is present

hursto75, 2017-03-14:
Companion build on Fedora 25 64bit, from source, errors

[bkelley@opentxbuild lbuild]$ cmake ../src
-- The C compiler identification is GNU 6.3.1
-- The CXX compiler identification is GNU 6.3.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Simulators library search path: /usr/local/lib/companion
CMake Error at CMakeLists.txt:98 (use_cxx11):
  Unknown CMake command "use_cxx11".

CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.6)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring incomplete, errors occurred!
See also "/home/bkelley/build/build22/opentx/companion/lbuild/CMakeFiles/CMakeOutput.log".
[bkelley@opentxbuild lbuild]$ su
Password: 
[root@opentxbuild lbuild]# ls
CMakeCache.txt  CMakeFiles
[root@opentxbuild lbuild]# cmake --version
cmake version 3.6.2

mpaperno, 2017-03-14:
I'm not sure what you're building.... Which instructions are you using?  Are you using CMakeLists.txt from the **root** of the source tree (not the ones inside any of the src folders)?

https://github.com/opentx/opentx/blob/next/CMakeLists.txt#L12

`use_cxx11` is a macro in https://github.com/opentx/opentx/blob/next/cmake/Macros.cmake#L45
Included from here: https://github.com/opentx/opentx/blob/next/CMakeLists.txt#L38

hursto75, 2017-03-14:
After doing a  git clone --recursive -b next https://github.com/opentx/opentx.git

cd opentx/
cd companion/
mkdir lbuild
cd lbuild/
cmake ../src

schwabe, 2017-03-14:
that is still wrong. Use the top level cmakefile and **not** the one of companion src dir. See https://github.com/opentx/opentx/wiki/OpenTX-2.2-Linux-Build-Instructions

closed 2017-03-14 by schwabe

mpaperno, 2017-03-14:
So yes, you're in fact trying to build inside the companion folder.  Back out one level and go from there.

https://github.com/opentx/opentx/wiki/OpenTX-2.2-Linux-Build-Instructions

```
$ cd opentx
opentx $ mkdir build-taranis-debug && cd $_
opentx/build-taranis-debug $ cmake ,,,
```

hursto75, 2017-03-14:
You are right... i was doing it wrong... Sorry I didn't read enough.

build-taranis-debug]# cmake -DPCB=X9D -DGVARS=YES -DLUA=YES -DDEBUG=YES -DCMAKE_BUILD_TYPE=Debug ../
 cmake finishes

make firmware

 23%] Built target firmware_translations
Scanning dependencies of target bootloader
[ 23%] Building C object radio/src/targets/taranis/bootloader/CMakeFiles/bootloader.dir/__/__/common/arm/stm32/f2/system_stm32f2xx.c.o
/bin/sh: arm-none-eabi-gcc: command not found
radio/src/targets/taranis/bootloader/CMakeFiles/bootloader.dir/build.make:62: recipe for target 'radio/src/targets/taranis/bootloader/CMakeFiles/bootloader.dir/__/__/common/arm/stm32/f2/system_stm32f2xx.c.o' failed
make[3]: *** [radio/src/targets/taranis/bootloader/CMakeFiles/bootloader.dir/__/__/common/arm/stm32/f2/system_stm32f2xx.c.o] Error 127
CMakeFiles/Makefile2:1522: recipe for target 'radio/src/targets/taranis/bootloader/CMakeFiles/bootloader.dir/all' failed
make[2]: *** [radio/src/targets/taranis/bootloader/CMakeFiles/bootloader.dir/all] Error 2
CMakeFiles/Makefile2:123: recipe for target 'radio/src/CMakeFiles/firmware.dir/rule' failed
make[1]: *** [radio/src/CMakeFiles/firmware.dir/rule] Error 2
Makefile:118: recipe for target 'firmware' failed
make: *** [firmware] Error 2

schwabe, 2017-03-14:
*arm*-none-eabi-gcc: command not found

avr and arm are different architectures. You need an arm gcc compiler.

hursto75, 2017-03-14:
Got it, to build firmware,
needed Installed Packages
arm-none-eabi-binutils-cs.x86_64                                       1:2.27-1.fc25                                          
arm-none-eabi-gcc-cs.x86_64                                            1:6.2.0-2.fc25                                         
arm-none-eabi-gcc-cs-c++.x86_64                                        1:6.2.0-2.fc25                                         
arm-none-eabi-newlib.noarch                                            2.4.0-9.fc25

mpaperno, 2017-03-14:
If you could contribute text or a patch to the current instructions for installing the  prerequisites on Fedora, that would be most welcome.

hursto75, 2017-03-14:
Exactly what I am working on, running down the dependencies now.

Brad

hursto75, 2017-03-14:
I am going to get this building then, i am going to test it it on a fresh
install then I will write up the process.

Brad

hursto75, 2017-03-14:
For make flash
errors:

[100%] Built target firmware
Scanning dependencies of target flash
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: No DFU capable USB device available
radio/src/CMakeFiles/flash.dir/build.make:57: recipe for target 'radio/src/CMakeFiles/flash' failed
make[3]: *** [radio/src/CMakeFiles/flash] Error 74
CMakeFiles/Makefile2:180: recipe for target 'radio/src/CMakeFiles/flash.dir/all' failed
make[2]: *** [radio/src/CMakeFiles/flash.dir/all] Error 2
CMakeFiles/Makefile2:187: recipe for target 'radio/src/CMakeFiles/flash.dir/rule' failed
make[1]: *** [radio/src/CMakeFiles/flash.dir/rule] Error 2
Makefile:210: recipe for target 'flash' failed
make: *** [flash] Error 2

mpaperno, 2017-03-14:
Do you have a radio plugged in @ USB and in bootloader mode?  What about `udev` permissions?  I forget if STM32 bootloader needs a driver on Linux... (I think not?)  Anyway this isn't really a build issue.

> I am going to get this building then, i am going to test it it on a fresh install then I will write up the process.

👍

hursto75, 2017-03-14:
No it isnt, I have it working, just need to test further.

hursto75, 2017-03-14:
To build under Fedora 25
As root:
dnf install git gcc svn qt qt-devel cmake patch xsd gcc-c++ SDL-devel phonon phonon-devel avr-gcc-c++ avr-gcc avr-binutils arm-none-eabi-binutils-cs arm-none-eabi-gcc-cs arm-none-eabi-gcc-cs-c++ arm-none-eabi-newlib.noarch avr-binutils avr-gcc avr-gcc-c++ avr-libc.noarch PyQt4 fox-devel qt5 qt5-qtmultimedia-devel qt5-devel.noarch qt5-qtwebkit-devel qt5-qttools-devel dfu-util

As user:

git clone --recursive -b next https://github.com/opentx/opentx.git

cd opentx/

mkdir build-taranis-debug/

cd build-taranis-debug/

cmake -DPCB=X9D+ -DGVARS=YES -DLUA=YES -DDEBUG=YES -DCMAKE_BUILD_TYPE=Debug ../

make firmware

make companion22

make flash

As root:

make install

schwabe, 2017-03-14:
I copied the package list to 

https://github.com/opentx/opentx/wiki/OpenTX-2.2-Linux-Build-Instructions#on-fedora

Thanks! The rest should be identical/not specific to fedora.

mpaperno, 2017-03-14:
Thaks @hursto75.   Are you sure about `qt5-qtwebkit-devel` being needed, and also what about `qt5-svg`?  Do you see icons in the Simulator of your build?

hursto75, 2017-03-14:
They were required in other Linux OS builds, so I included them. I am not
sure if they are needed or not for this build. I develop and RPM for this
and other RPM based installations when 2.2 is released.

Thanks,
Brad Kelley

adelton, 2017-08-20:
@hursto75, I have built 2.1 and 2.2 version rpms of Companion at https://copr.fedorainfracloud.org/coprs/adelton/opentx-companion/ and https://copr.fedorainfracloud.org/coprs/adelton/opentx-companion-2.2/.

The .spec's and .src.rpm's with the patches are included in the build result so it should be easy to see how the build was done.