opentx/opentx #4618: Question: EEPROM vs EEPROMS and FIRMWARE vs FIRMWARES

lshems, 2017-03-14:
Is there a deliberate change in the names of these SDcard folders from 2.1 to 2.2, and if so, why??

It is just confusing if you switch regularly from openTX versions on the radio.

3djc, 2017-03-15:
Yes deliberate choices, to reflect better the purpose of those folders (BMP to IMAGES to,...)

closed 2017-03-15 by 3djc

lshems, 2017-03-15:
OK, its not a big deal.

But I still don't see a reason for a difference in approach for models with S and firmware without S, where it was with S before. There will be multiple firmwares in the firmware folder, as well as multiple Eeproms in the Eeprom directory, at least if you do a proper management of your setups and versions, and want to be able to return to a previous state. So why those had to change is beyond my comprehension.