opentx/opentx #4622: Dropping mixer lines shifting one line

lshems, 2017-03-14:
when dragging and dropping a mixer line, sometimes it gets dropped on the previous mixer line instead of the mixer the black line is put after.

elecpower, 2017-03-16:
Noticed this behaviour some time ago when fixing the drag and drop before line 1 issue. My testing showed that when moving the mouse very slowly across mixer lines there is a very small shift in the black line. That small shift is enough to affect which line the dragged line is dropped onto.
Since it was a minor annoyance and once I had worked out where to drop didn't bother to pursue.
Not sure if it is an inherent QT issue, related to the padding (if still exists) between mixer lines or even OS dependent.

projectkk2glider, 2017-05-05:
Which version? Is behavior in 2.1 and 2.2 different - they use different QT verisons?

PR #4919 (2017-06-10): [Companion] Inputs and Mixes lists formatting & DnD fixes:
  * Fix/improve margin/padding issues, consistent between inp/mix views, better handling of x-platform and older Qt (closes #4599, closes #4341);
  * Allow dropping onto lines instead of only between them;
  * Fix dropped data always going to line above the actual drop target (closes #4622);
  * Force DnD _move_ operations between windows to _copy_ data instead (fixes deletion of any selected line in target window, and source line was never actually removed but looked like it was);
  * Add alternate-row coloring for each Input/Mixer gr [... cut]
commit 3f5a3995 (2017-06-10): [Companion] Inputs and Mixes lists formatting & DnD fixes: (#4919)

* Fix/improve margin/padding issues, consistent between inp/mix views, better x-platform and older Qt (closes #4599, closes #4341);
    * Allow dropping onto lines instead of only between them;
    * Fix dropped data always going to line above the actual drop target (closes #4622);
    * Always force DnD move operations between windows to copy instead (fixes deletion of any selected line in target window, and source line was never actually removed but looked like it was);
    * Add alternate-row coloring for each Input/Mixer g [... cut]
closed 2017-06-10 by bsongis