opentx/opentx #4624: TX Antenna Warning can't be turned off

lucerne86, 2017-03-15:
I recently ripped off the antenna of my Taranis X9D+ by accident when handling my racecopter. Now i wanted to use the Taranis to play Simulator games on my PC but i can't, i get the "TX Antenna Warning" every 5 seconds and it can't be turned off.

I understand that the main purpose of this TX is to fly real aircrafts and not PC sims but anyway there should be a way to dismiss this warning or turn it off in the settings.

lichtl, 2017-03-15:
Have you tried to disable the internal TX module?

lucerne86, 2017-03-15:
Wow! Thank you million times for your quick response! You solved my problem 👍 :-D
I was already thinking about making my own build of OpenTX to solve that..

Have a nice day!

closed 2017-03-15 by lucerne86