opentx/opentx #4628: TaranisQ X7 - No Wizards starts for new plane N364

dklueh79, 2017-03-16:
Setting up a new plane, the wizard for a new model doensn't open. The lua wizards doesn't work on manual execute.

3djc, 2017-03-16:
There is currently no wizard for x7

closed 2017-03-16 by 3djc

dklueh79, 2017-03-16:
Ok, then the sd-card content shouldn't contain the wizards at all

3djc, 2017-03-16:
Unless I'm mistaken, it doesn't

kilrah, 2017-03-16:
It doesn't, you probably took the x9 zip or copied your card from an x9 that had it.

henriquedesousa, 2017-08-16:
Is it technically possible for someone to port it to the QX7, taking into account the reduced screen?

kilrah, 2017-08-16:
Anything can always be done, it's only the amount of effort that varies.
In this case since the wizard was made to use all the nice graphics capabilities of the X9 screen (that the X7 doesn't have) all the UI basically needs to be redone differently. 
Nobody has shown interest in doing that job so far.

henriquedesousa, 2017-08-17:
I just found that OpenTX companion has the same wizard, so we're good for now. Many thanks.