opentx/opentx #4629: RSSI does not reset to 0

kcunreal, 2017-03-16:
In OpenTX 2.1.9 the RSSI Value does not reset to 0 when battery is disconnected from multi rotor.  I was trying to configure a logical switch that would reset my timer when RSSI value would go to 0 like after unplugging the battery.  This worked fine in version 2.1.6 (and I switched back to from 2.1.9).

bitogre, 2017-03-17:
This is an issue that was first discussed in much detail on the [FliteTest Forums](http://forum.flitetest.com/showthread.php?33214-Taranis-RSSI-reset-when-unplugging-battery).  

Really, the needed functionality is a way to detect a link loss to the receiver.  In 2.1.6, you could use RSSI for this detection.  While the change to have telemetry hold most previous values after loss of link does make a lot of sense, ether it should not apply to RSSI (which is in effect a link quality metric) or there needs to be added some other means to detect a loss of link to the receiver.

Mow06, 2017-03-17:
logical switch a>x rxBt (or a1) 1v goes false when telemetry lost.
All LS testing telemetry values go false when telemetry lost.
Works for me.

bitogre, 2017-03-17:
@Mow06 Are you saying the test is that the LS goes false?  So if I have one logical switch a>x rxBt (or a1) 1v and another logical switch a<x rxBt (or a1) 1v (which normally should never both be false at the same time), when I see both false, that is how I can tell there is a link loss?

Mow06, 2017-03-17:
Should be so. I just use a>x going false.

kilrah, 2017-03-22:
More relevant discussion in #4663.

As said any telemetry based LS is false if no telemetry is received, so say a>x RSSI 0 will go false with no telemetry.

closed 2017-03-22 by kilrah