opentx/opentx #4631: [X9D] GPS coordinates are sometimes badly displayed

bsongis, 2017-03-17:
[image]
[image]
[image]

sirPerna, 2017-03-17:
only values <1.0 affected.

3djc, 2017-03-17:
Could you post bigger pictures, can't see with those :)

bsongis, 2017-03-17:
ah yes

bsongis, 2017-03-17:
[image]
[image]
[image]

3djc, 2017-03-17:
With the coordinates taken from the fpv screenshot
N364 DMS:
[image]

N364 NMEA
[image]

closed 2017-03-18 by 3djc

sirPerna, 2017-03-18:
@3djc Did you fix anything becase it's still not working here. The CRSF sends Latitude first. 

[image]

3djc, 2017-03-18:
What version are you using ?

sirPerna, 2017-03-18:
2.2.0N364 the latest it downloaded yesterday.

kilrah, 2017-03-18:
Well we didn't even know you were using CRSF telemetry... We thus checked with FrSky obviously.

sirPerna, 2017-03-18:
Sorry for the bad report in this case. Setup is TBS CORE PRO - Crossfire - Taranis. I can share a test firmware for Crossfire so no additional hardware required then a Crossfire TX and a OpenTX remote.

3djc, 2017-03-18:
If I input your values with 364, i get the screens above (mine) that are correct.
If I used N362, I get your screen :
[image]

Can you confirm on the radio you are running N364, because it doesn't look like it

kilrah, 2017-03-18:
> 2.2.0N364 the latest it downloaded yesterday.

Then you probably downloaded **companion** N364, but didn't download / flash N364 **firmware**.

bsongis, 2017-03-18:
@sirPerna would you tell us which firmware version you see in the VERSION screen on the radio?

sirPerna, 2017-03-18:
[image]

bsongis, 2017-03-18:
ok, we have a bug for sure

bsongis, 2017-03-18:
bug confirmed even in our current 'next' branch

bsongis, 2017-03-18:
the GPS parsing code is here:
https://github.com/opentx/opentx/blob/next/radio/src/telemetry/crossfire.cpp

```
if (getCrossfireTelemetryValue<4>(3, value))
        processCrossfireTelemetryValue(GPS_LATITUDE_INDEX, value/10);
```

bsongis, 2017-03-18:
it returns an uint32_t which is not good here

bsongis, 2017-03-18:
@3djc I take it, testing in progress with @sirPerna

sirPerna, 2017-03-18:
Tested. Thank you!

commit 99472b4f (2017-03-18): Fixes #4631

Tested by @sirPerma
commit 87685cf8 (2017-03-18): Fixes #4631 - Same problem was happening on ATTITUDE (16bits signed value)

Tested by @sirPerma