opentx/opentx #4636: No new Firmware available thru companion N365

dklueh79, 2017-03-19:
Neither for X7 nor Horus you could download the new firmware. The last is the 364 - check for nightly is set in setting.
[image]

dklueh79, 2017-03-19:
[image]
[image]

3djc, 2017-03-19:
Fixed

closed 2017-03-19 by 3djc

dklueh79, 2017-03-19:
Very fast! Thx!!!!

kilrah, 2017-03-19:
Things get built one after the other, so become ready one after the other, normal.

We'll be changing to hold off the notification until everything's ready for the future.