opentx/opentx #4637: Missing Qt translation

LapinFou, 2017-03-19:
Hi folks,

When closing Companion without saving the modifications, the popup windows got some fields in English.
I didn't see them with Qt Linguist. A guess there is a Qt "tr(xx)" missing somewhere.
[image]

3djc, 2017-03-19:
They are qt built in messages linked to your system language, cannot be changed.

kilrah, 2017-03-19:
Tey can, reference:
http://stackoverflow.com/questions/31533019/title-qt-how-to-translate-the-buttons-in-qmessagebox

LapinFou, 2017-03-19:
But, I'm running a "French Window 10 famille" version?
So, something is not right.My "préférences inguistique" is setup to "français(France)".

3djc, 2017-03-19:
That's really odd! There is a magic trick you can do in case like that. Let me try it : @mpaperno!

kilrah, 2017-03-19:
See my post above!

LapinFou, 2017-03-19:
Same thing in the Wizard.
[image]

On the French forum, somebody reported on MAC than the "close, minimize, et maximize" buttons (top left of the windows) are not translated either.

3djc, 2017-03-19:
for the same reasons

LapinFou, 2017-03-19:
Yep, probably. 😄 

Here is the link, just in case: http://frskytaranis.forumactif.org/t5037-opentx-2-2-0-rc13-n365#58134

3djc, 2017-03-19:
It is the case on windows too btw ;)

LapinFou, 2017-03-19:
In fact same issue on Windows. It is not only Mac OS dependent.
[image]

3djc, 2017-03-19:
You will find it in many many places, where system dialogs are used

LapinFou, 2017-03-19:
OK. So, I will wait @mpaperno feedback.

LapinFou, 2017-03-19:
This one is a missing Qt translation:
[image]

kilrah, 2017-03-19:
The one on the pots does have the tr(). Did you update the .ts file?

LapinFou, 2017-03-19:
I didn't see it in the .ts file generated by @3djc . Could you (or @3djc) generate a branch with the latest up-to-date French .ts file? I don't have the flow installed on my computer (just Qt Linguist).

kilrah, 2017-03-19:
Weird, it's actually already there and translated :\

[image]

LapinFou, 2017-03-19:
Badly translated, it should be "Double-clic **droit** pour recentrer."

mpaperno, 2017-03-20:
> There is a magic trick you can do in case like that.  

LOL...  obviously I should ignore you more often!  :)

Well Andre gave the link (2nd answer there is correct), but essentially Qt comes with a bunch of `.qm` files for its internal strings like buttons and other UI elements.  In a downloaded Qt SDK they're in the `Qt/<x.y>/<toolchain>/translations` folder, broken up somewhat by module (most/all stuff we need should be in `qtbase` and maybe `qt`).  They can be packaged with an app same way as custom translations (either qm files in a folder with the app, or embedded into a resource file like we do now.)  Multiple translation files can be used in an app at once: http://doc.qt.io/qt-5/qcoreapplication.html#installTranslator

The minimize/maximize buttons (as in one of the screens above) are controlled/drawn by the OS.  However we already learned recently that Windows hasn't updated the MDI sub-window design in several iterations, so that could be the reason why the main window min/max buttons are translated, but not the sub-window ones.  I'm not sure if installing the Qt translations will help in this particular case, but I'm guessing not.

3djc, 2017-03-20:
Well, I was calling on the most knowledgeable person I know around Qt :) I did read the link on how you can replace those default translations by your own, what I did not understand it why on a common language like french, there not there 'by default'. Maybe we should try packaging those default translation, it surely make more sense than try to redo all those ?

3djc, 2017-03-20:
@LapinFou , here is an up to date file
[companion_fr.zip](https://github.com/opentx/opentx/files/854112/companion_fr.zip)

LapinFou, 2017-03-20:
@3djc : thanks for the file. I will have a look.

LapinFou, 2017-03-21:
@3djc 
I push the latest translation #4657 according your .zip archive.

LapinFou, 2017-03-23:
Hi folks,

In order to track the progress between the RC13 and the RC14 (N366) none of the previously issues named in this ticket have been fixed.
In the RC14 (wahouuuu, the Companion simulator is getting better and better. You are greats!!! 💯 ), I discovered few more missing tr():
[image]

mpaperno, 2017-03-23:
The top 2 are brand new as of yesterday... what's taking you so long to translate them?   😛 

The last one (file type name) is actually hidden in a macro (in constants.h) and right now is not translatable... will put that on the "to be fixed" list.

I'm taking care of getting the Qt translations included and also make it easier to preview new translations within Companion/Simulator while working on them (w/out re-compiling).

LapinFou, 2017-03-23:
OMG!!! I'm one day late!! 😓 
Understood for the file extension.

Being able to test Qt translation would be very very handy for translators (helpful to get the context where the text is used). 👍 

Since, I don't have any dev environment (don't want for now, only got a standalone Qt Linguist), could you kindly send me the latest french .ts file?

mpaperno, 2017-03-23:
"Latest" is relative  😄   We should probably just regen/commit all the .ts files in `next` 

BTW you do not need Qt or any compiler installed to generate .ts files -- Linguist comes with the `lupdate` utility which is what is used to generate .ts from code.  (You also use the included `lrelease` utility to generate the final binary `.qm` file from `.ts` file.)  You do need the full OTx source code available, but I imagine you have that anyway (if not, you should because it is much nicer for translating to see source code reference).

I think you can run both utilities from a menu in Linguist, but also simple from a command line.  E.g:

`lupdate </path/to/code> -no-obsolete -ts </path/to/file.ts>`

`lupdate c:\OTx\  -no-obsolete -ts c:\OTx\companion\src\translations\companion_fr.ts`

You will see some warnings but that is OK.  The final output should be something like:

```
Updating 'c:/OTx/companion/src/translations/companion_fr.ts'...
    Found 2263 source text(s) (176 new and 2087 already existing)
    Removed 153 obsolete entries
    Same-text heuristic provided 22 translation(s)
```

LapinFou, 2017-03-23:
Thanks for those nice information. Unfortunately, I installed Qt 5.8.0 on a machine, create an archive with only required .exe/.dll, then uninstall it.
So, in my handmade Qt Linguist v5.8.0, I got only the **lrelease.exe** CLI.
It would be so nice, if you could send me the missing .exe through a ZIP archive.

schwabe, 2017-03-23:
@mpaperno you add a short "How to translate OpenTX page" in the wiki based on that information? I will then play dummy user and test it for German ;)

kilrah, 2017-03-23:
I think there's already a page, could be added to it

LapinFou, 2017-03-23:
@schwabe 
Too easy for you, most of the missing tr() have been found... 😉

schwabe, 2017-03-23:
@LapinFou I was more thinking about a five to ten step tutorial for using seeking to translate Companion into their language.

mpaperno, 2017-03-23:
> I think there's already a page, could be added to it

I thought so too but couldn't find it on the GH Wiki.  Is it in some other docs?

LapinFou, 2017-03-23:
@schwabe I understood. I was just joking. 😃 
Concerning the wiki, it could be nice to have a standalone Qt Linguist archive. Here is my handmade version (as explained previously, some executable are missing).
[Qt_Linguistv5.8.0.zip](https://github.com/opentx/opentx/files/865505/Qt_Linguistv5.8.0.zip)

kilrah, 2017-03-23:
Possible it was removed becasue so outdated it had to be rewritten from scratch anyway

mpaperno, 2017-03-23:
I can't believe there's still no easy way to d-load Linguist separately.  Though it is included with each Qt SDK install (using their [online/offline installers](https://www.qt.io/download-open-source/#section-2) ) 

In the installer component selector, choose any desktop OS version (eg. on Windows `MinGW*` or `msvc*`).  After install Linguist will be available in the usual [Start|Applications|Programming] [menu|folder] (depending on OS).

[image]

It also forces you to install Qt Creator as part of the package.  If disk space is an issue, after install you could delete everything except the `Qt\<x.y>` folder (where "x.y" is Qt version) and still keep using Linguist & tools (it is also possible to "extract" just the few bits actually needed for Linguist to run, like @LapinFou did).

@LapinFou Which compiler (MSVC? Versions?) did you build your Linguist with?  Whatever I send you would need to match.  One way to get the missing bits would be to install the matching SDK (as above) and just get what you need before uninstalling (lots of download though...).

kilrah, 2017-03-23:
Can find some old releases: https://www.linux-apps.com/content/show.php/Qt+Linguist+Download?content=89360

LapinFou, 2017-03-23:
@kilrah I tried the old release, but few things was not working well with the generated .ts file.

@mpaperno I don't remember which MSVC version was installed. 2 solutions:
1. You have a look in my .zip file → [Qt_Linguistv5.8.0.zip](https://github.com/opentx/opentx/files/865505/Qt_Linguistv5.8.0.zip). Then, you make the "full" handmade version. This version (if this is legal), could be downloaded somewhere int the Wiki. Thus, others translators can be "up and running" in no time.
2. I will install again Qt on my PC and make myself a 5.8.0 standalone version.
Since, I can easily guess than you are very very busy, I'm not sure you will have sometime for the 1st solution.

mpaperno, 2017-03-23:
Qt translations DO handle the MDI subwindow controls  :)

[image]

mpaperno, 2017-03-23:
@LapinFou I packaged up a version of Linguist from my local Qt SDK install.  You could see if this works for you.  Just unzip somewhere and run `linguist.exe`.  If it works we could make this available somewhere from the OTx site/wiki (this one is Windows-only, obviously).

http://max.wdg.us/OpenTX/QtLinguist-5.7.1-msvc2015.zip

I see `lupdate` does not have a menu option in Linuguist so indeed it must be run from a command line.

LapinFou, 2017-03-23:
Just tried. Qt Linguist doesn't work, I got this error message:
[image]
I think the sub-folder **platforms** is missing. In this sub-folder, the file **qwindows.dll** should be present.

I tried the CLI. It works!! 👍
```
D:\GitHub\Qt Linguist>lupdate.exe D:\GitHub\OpenTX -no-obsolete -ts  D:\GitHub\OpenTX\companion\src\translations\companion_fr.ts
Scanning directory 'D:\GitHub\OpenTX'...
D:/GitHub/OpenTX/companion/src/firmwares/opentx/opentxinterface.cpp:732: Qualifying with unknown namespace/class ::OpenTxFirmware
[...]
D:/GitHub/OpenTX/radio/src/switches.cpp:936: Parenthesis/brace mismatch between #if and #else branches; using #if branch
D:/GitHub/OpenTX/radio/src/switches.cpp:724: Unbalanced opening brace in C++ code (or abuse of the C++ preprocessor)
Updating '../OpenTX/companion/src/translations/companion_fr.ts'...
    Found 2317 source text(s) (0 new and 2317 already existing) 
```

mpaperno, 2017-03-23:
> I think the sub-folder platforms is missing. In this sub-folder, the file qwindows.dll should be present.

You're right... and it is present I just put it in the wrong place... d'oh!  Drag the 3 folders inside the `plugins` folder up one level (same directory as `translations`).   LMK if that works.

> I tried the CLI. It works!

👍

LapinFou, 2017-03-24:
@mpaperno I just tried. Moving the folder **printsupport**, **platforms** and **imageformats** from the __plugins__ directory to the root one di solved the problem.
This standalone version is working perfectly on my Win10 64bits PC.
Thanks! 👍

mpaperno, 2017-03-24:
Thanks for letting us know!  It now has an official home:  http://www.open-tx.org/2017/03/24/QtLinguist-download   😹 

Still need that translators instructions page for the Wiki though...  which also got a reorganization BTW (so it's all ready for new content 😉 ).

LapinFou, 2017-03-24:
I'm quite busy right now, but I will try to create an up-to-date "Translation tutorial".

I generated the latest french .ts file. I got one question:
In the section **Multiprotocols**, the translations in the file _opentx\companion\src\multiprotocols.cpp_ are only for Companion? Or are they also used for the OTX firmware itself? Just to know if a can or not use some foreign characters.

schwabe, 2017-03-24:
Only companion. Radio has its own translation mechanism that is not compatible with companion.

LapinFou, 2017-03-24:
Thanks. Good to know.

PR #4676 (2017-03-25): [Companion] Translation system improvements.
  * Introduce new Translations class to centralize all related functionality;
  * Add ability to search multiple locations for .qm translation binaries (local folders, resources, system paths);
  * Add ability to reload translations dynamically w/out application restart;
  * Detect and use system language settings to load correct translations by default;
  * Improve support for loading Qt translations (if available);
  * Look for and include pre-built Qt translations with other translation resources (fixes #4637);
  * Generate translations.qrc dynam [... cut]
closed 2017-03-25 by bsongis