opentx/opentx #4638: Design N364 Radio Output Channel Pointers  should be coloured like V2.19 was

hrenz, 2017-03-19:
Hy,
to see the channel pointer position better at Radio Output Simulator 
they should be coloured like V2.19 was, blue or green or red.

Or can I set it somewere?
Win7 prof openTx V2.20N364

hrenz, 2017-03-20:
Channel  Position Pointer not coloured, bad to see.
[image]

mpaperno, 2017-03-21:
Would this be better?  

[image]

mpaperno, 2017-03-21:
Or actually darker color on bottom, I think... ?

[image]

kilrah, 2017-03-22:
Prefer the 2nd :)

closed 2017-03-22 by kilrah

hrenz, 2017-03-22:
2nd looks fine 
thanks

PR #4669 (2017-03-22): [Simulator] New skin for channel output indicators. 
[image]

Could someone try this on a high-def display to make sure it scales properly?  TIA.

(closes #4638)
commit dc00757c (2017-03-22): [Simulator] New skin for channel output indicators. (Fix #4638 ) (#4669)