opentx/opentx #4639: OFF- switch without delay for Horus

1Sven, 2017-03-19:
The OFF- switch must be hold for very long time before the TX goes down.
It is not important it is annoying. No one can switch off unconsciously, the lid over the switch reliably…
Please make the push for “OFF” shorter...

marcogpw, 2017-03-19:
I like the delay, almost impossible to switch off your Horus by mistake. A user selectable delay of 1, 2, 3 or 4 seconds could be an option for our "always in a hurry people" 😜

schwabe, 2017-03-20:
I did not implement that feature or had anything to do with this decision. But this is one the decision of "does the default hurt the users that don't like the default enough to add the complexity/mistake potiental to make it configurable". And in this the answer is that all people can hold the power off 5s

1Sven, 2017-03-20:
Of course every body can hold it. But why? On the Taranis it works fine without any thing, of course it's hardware.. Horus has a additional security flap, it is absolute ok. May be, not for the time just, but I hope for understanding that a simple operation is a good way. Please, take a heart for the "always in a hurry people" ... The push ON works great since last version, why not power off...

3djc, 2017-03-20:
Because powering off is a much more dangerous operation than turning on ?

marcogpw, 2017-03-20:
@3djc that's what I think as well: better be safe than sorry! The OpenTX way of switching off is also better than the FrSky way (IMHO)

RealTadango, 2017-03-21:
I have tested a long time and a reduction from 3 to 1,5 second is MUCH better. This makes the startup / shutdown speed IDENTICAL to my Spektrum DX18. 3 second is TO LONG and is really annoying. 1 second is to fast. 1.5 seconds is the perfect time (for me).

So PLEASE change it, many have requested it.

mihauOne87, 2017-03-25:
Delay options for hurry people is a great idea !

vitas, 2017-04-27:
yes, please reduce the delay to 1,5 or max 2 sec

projectkk2glider, 2017-05-05:
I would not mind the shorter shutdown time (in fact I would probably like it), but are we **really sure** it won't cause any problems or safety issues?

RealTadango, 2017-05-05:
It will be exactly the same as my DX 18 with 1,5sec. I am using it for a few months now and it works perfect. But that is just my opinion ;)

kilrah, 2017-05-05:
I'm still for a configurable delay since different people have very different opinions about it. 
We should also harmonize power on (currently we support 3 radios with pushbutton, x9e has a power up delay but for some reason horus and x7 don't)

AABatteries, 2017-05-06:
@3djc thanks for referencing my issue and sorry for creating a duplicate. Configurable shutdown delay would be a great enhancement for X7 for sure

3djc, 2017-05-06:
Sure, but given the position of the swicth on x7, there is a real risk of error with trims, so I personally doubt we will allow it short on x7

markran, 2017-07-06:
I would also really like to have this be configurable. Default it to 1.5s but please also have a no delay option.

lshems, 2017-07-07:
Why mix it up with trims? Really funny. 

I do understand the risk of accidentally shutting down the transmitter, but could it be done in a sort of safe way directly, for instance by having it work direct when holding the menu button down?

kilrah, 2017-07-07:
> Why mix it up with trims? Really funny.

There are several people who lost models because they turned their X9 off while trying to find the trim with their finger and reaching too far. Same would happen if only a single short press was enough to turn off the radios with pushbuttons.

1Sven, 2017-07-07:
It is on Horus a real catastrophe..., first it is nearly not possibele to mouve the strap-holder (have thickness fingers) and then is the about 5s a very long time. Often I must press again while the figer was not correct pressing.  I think a little hardware to bridge the knob is the solution. I will think about it now..., I have heared for software it isn't possibele before 2.3 while eeprom- change needed and I must realy do something. If I have a solution I will tell it...

lshems, 2017-07-07:
On taranis on of is a slider like trim. On x7 trim is slider and on/of is push.

So to me it's still seems funny for someone to mix them up. 

It seems like comparing apples with pears here.

schwabe, 2017-07-07:
@lshems  People did confuse them, hit them by accident. So only that you cannot imagine that someone else does. Every used a remote with gloves in winter? It is much easier then to hit the wrong switch for example

RealTadango, 2017-07-08:
The Horus had a protected lid over the button andere does not have an issue. Just adjust the time for the Horus....

1Sven, 2017-07-08:
Oh really ? I was thinking there is a Eprom change nessesary! Of corse, if you can make the time per software shorter, I don'nt need to frickle on my Horus... Can I hope that it will avilable on the 2.21?

lshems, 2017-07-09:
@schwabe Still funny to me for an X7. But the comment was not about that. Basic comment was that for those really wanting a direct switch off, some combination of buttons pressed could be a solution. Like menu pressed while pressing on/off would switch of directly.

I'm not in need of any solution, just suggesting for those who are.

OhmEye, 2017-08-11:
I don't understand the objections to making the delay configurable. Leave the default at 5 seconds and people can choose a shorter delay and assume whatever risks, real or imagined. On my X7 I find it nearly impossible to accidentally touch the power switch, it's recessed and beneath the strap balancer. My fingers don't reach it when using the trims, and it's not like even a quick press will hurt anything, it takes some deliberation to press and hold for even one second, 3+ seconds can be very annoying to do often. My 2 cents. Honestly it did not occur to me I wouldn't be able to change the delay until I went looking for where to do it and searching led me here.

1Sven, 2017-08-13:
It is a disaster. I have the Hours. For easy using i have cut the flap. i have never seen a switch of that was so katastrophal...

raphaelcoeffic, 2017-11-20:
I found it annoying as well, and basically just changed the define in the code to 1s, works good for me.

1Sven, 2017-11-20:
Oh, realy? I was thinking about EEPROM change mus be? Is it avilable in the next 2.2.1? Perfekt Solution.. Thanks!

RealTadango, 2017-11-20:
No the 1s timeout is only possible when you build yourself..... Unless the team accepts it as a solution.

raphaelcoeffic, 2017-11-20:
@RealTadango but we could probably make it a build parameter, so that it can be applied easier. At the moment, I always have to go change it in the code every time I change the branch, which is kind of annoying...

1Sven, 2017-11-20:
There are already many who would like that and the solution is tangible. How can the team be convinced?

1Sven, 2017-11-20:
HI, Hi…  das Team…

 

was ist eigentlich mit dem Ausschaltverhalten nun; ich denke, das ist gefixt?

 

Sven

 

Diese E-Mail wurde von Avast Antivirus-Software auf Viren geprüft.
https://www.avast.com/antivirus

raphaelcoeffic, 2017-11-20:
@1Sven I just noticed: it is assigned to 2.2.X milestone, and I heard that a configuration parameter will be added for that. However, as it requires an eeprom change, it will first be added when new eeprom changes are made (next major, 2.3?).

capitaineflam25, 2017-12-19:
I use 1s on my X7 build : i prefer it to the stock 3s delay.

I'm curious, why is there a 1s delay before powering up ? Anyone observe an unwanted power on an a X7 ?

raphaelcoeffic, 2017-12-20:
@capitaineflam25 yeah, I heard this is done to protect you from discharging your battery when stuffing your X7 directly into your backpack.... the same applies to the X10 (and maybe all TX with a simple button), AFAIK.

capitaineflam25, 2017-12-20:
Thanks Raphael

1Sven, 2017-12-20:
Oh no! Please not 3s ore more for the X12, it'll shut up .. I think you should be able to register your own value.
> Thanks Raphael for the (for me) perfect pre-release version! <

raphaelcoeffic, 2017-12-20:
I guess we can have a lot more options in 2.3 to better reflect user preferences (ex: for those who don't stuff their radio in their backpack, or are not afraid of shutting it down per inadvertence). But for now, all we have are macros in the code.

CraigHB, 2017-12-21:
I have the same issue with an X7.  I would really like control over the delay.  Also with 2.2.1 the haptic, does not adhere to tx settings.  Haptic always fires on turn-off and turn-on even when haptic is set to quiet.

vinito, 2017-12-22:
I really don't understand why I see a sort of "push" from some people to not change (impose?) and leave be such a long power-off delay. Just make it configurable and those who want that can set it that way, i.e. their own... _preference_, and make it so you can set it to a half second for those on the other side of the fence, i.e. their own... _preference_. Make it user-configurable and folks can be super-duper-safe and set it to 15 seconds if you want! I thought the whole idea of OpenTX is so you can configure your whole radio experience to suit _your own preferences_. Just the simple fact that some folks have large, heavy camera rigs and some might only have tiny whoops can easily change what you want or should set on your radio for safety reasons. 
For that reason alone users would want different settings for the power off delay just for different models rather than a global setting - whether a per-model setting would be possible or not is another story, but the argument is still valid nonetheless. A global setting to suit individual preference would be better than having it fixed, especially since **it's excessive** for most people in the first place.
Whatever the default delay is feels way to long to me for the stuff I'm flying - some say it's 3 seconds but it feels like 5 to me. For me, a single full second would be plenty adequate **and safe** to avoid accidental shutdown. A second _sounds_ short but it's longer than most folks think much about. 1-1/2 sec. as mentioned before would be more reasonable, but picking your own preference would be much better as that obviously varies between people for several good reasons.
I also don't see how you could mistake trim for the power button, even with gloves on or maybe even especially with gloves on. It's not easy to reach the power button down in the hole (I'm speaking of my X7 since that's what I've got) and at least 5 times easier to reach trims protruding above the face, and even a 1 second delay would make it _magnitudes_ less likely to mistake one for the other. Having said that, if you are concerned about mistaking trim for power button, then by all means set your power-off delay to whatever you like! Heck, I'm all for being safe and don't want to see anyone be injured, but I'm also all for having options and being responsible for that on my own. FREEEEEDOMMMM!, hehe. I'm serious though.
I hope this option gets implemented soon. Since I power my transmitter on and off maybe a couple dozen times within an hour or two while I'm setting up a new quad, and the long delay is annoying _each and every time_, it's quite a significant negative for the transmitter which otherwise has almost zero negatives otherwise and certainly no significant problems in my estimation.

kilrah, 2017-12-22:
Cut the drama, **it has been said it would be made configurable**.

It also has been said that **it requires a storage update**, which means **it needs to wait for the next major release**, which means **dont expect it before some time mid next year**.

vinito, 2017-12-22:
Didn't intend to come off dramatic. I just wrote that to present my thoughts, not to rant. It's the old missing face-to-face internet communication problem.
Since it has also been said "... doubt we will allow it short on x7" (just picking one, not calling anybody out), and newbie me is not versed on what the decision makers are really planning, I just thought I'd throw my thought out there for what it's worth. I tend to be verbose - apologies.

CraigHB, 2018-02-25:
Saw that, was just wondering if that changed since it was mentioned some months ago.

3djc, 2018-02-25:
Since the technical issues underlying that statement haven't changed, there is no reason for this to change

marcogpw, 2018-02-25:
Stop nagging about these little things, let the team work and I was wondering how much the complainers are donating to the OpenTX team ?? The team is doing it in their spare time, it’s not their full time job and they don’t work for FrSky ...

1Sven, 2018-02-25:
It's just a matter of if and when the programmers want that.
OTX is always what is important from the point of view of this group. That's why the programmers have to be convinced! That's why it's also important to keep repeating that programming is not just about complex, complex things, but also simple features that the average user likes. For this case has already been communicated that it will be implemented. It will be soon ... (Google Translate)

vinito, 2018-03-01:
A couple questions... none specifically on-topic but rise to the surface based on what has been posted here.
**1)** Is there another place anywhere where users (as opposed to developers) of OpenTX should be voicing complaints or suggestions? The mood of the contributors' and others' replies here are rather aggressive in response to fairly benign user comments, which leads me to think that this is not the place for feedback, but rather for contributors only to be posting their own internal communications. If this is the case, then maybe rather than continuing to field arguments from ignorance, such as my own, you can direct user feedback to the proper place. If this is not the case, then I don't really have anything constructive to say about that I guess.
**2)** Is there a "recipe" somewhere (I've looked a little with no success) to show how to build my own firmware such as others apparently do? This would be interesting to explore and if made more accessible, maybe more people (including myself) could contribute a thing or two. Besides being able to customize OTX to actually suit my own needs, it could lighten the load on existing devs if there were more in the pool. It seems odd that some folks can make their own solo builds to solve an issue while others say it is too fundamental to easily change until a major release is complete. If nothing else, knowing what it takes to roll your own build might help folks like me understand why.

Thanks in advance for any advice about 1) OpenTX user suggestions and/or 2) instructions for how builds are made.

kilrah, 2018-03-02:
It is the correct place for well-thought comments. If you read the thread, you'll see the reason some developers start getting annoyed is because some people keep complaining when the issue was already accepted and promised to be handled.

For building see the Wiki. https://github.com/opentx/opentx/wiki/Compiling-OpenTx

RealTadango, 2018-03-02:
I think a discussion can be held better at the chat: https://opentx.rocket.chat/channel/OpenTX_General, complaints also. This place should only be for bugs / improvement requests after consulting with the team.

lshems, 2018-03-02:
@vinito 

The tone on this github is sometimes not one of peace and harmony, let alone patience and respect.

You are not the first to notice. 

I also suggested some changes in the way of working to prevent it. 

The effort to change is deemed to high for no gain in code output. That's my analysis anyway. 

So we just go on like this, and have some nasty posts every now and then. 

For me, the result is that I stopped posting most requests or suggestions I have. I live with OpenTX as it is, and that's it.

Pity, but hey, I am not missed here.

P.s. It may look like I do not valuate the work of the developers. I do. If I do not add this phrase, I will get blamed for that again.
P.p.s. I may sound bitter. Perhaps I am. I just don't get why it must be like that. Must be me.

vinito, 2019-01-05:
_"It also has been said that it requires a storage update, which means it needs to wait for the next major release, which means dont expect it before some time mid next year"_

So, not a peep for the mentioned year.
Sometime the following year then (this year) maybe? Since some guys have done their own builds to change the delays for themselves, I would have guessed that fixing it would be a trivial thing for a _team_ of devs., especially since OpenTX added a _power-on_ delay to the stock firmware almost immediately. Maybe the "team" isn't as big as I imagine? I admittedly know not how a dev. team structure works.

I also know not much about making my own build. Maybe this is all a good thing as I'm beginning to become inspired to learn something new and figure out how to do just that.

Anyway... So yeah, for my purposes, even after a year I still find the delay as annoying as ever. It isn't wearing off.
Just so ya know.

kilrah, 2019-01-06:
Changing it to be 0 seconds for everyone is easy, but as said we won't do that since we deem it unsafe. For us to change anything we will make it user-adjustable so that'd people who want 0 can set it without impacting others, and adding settings requires a major version change. 

The team size varies since people contribute (or not) on a voluntary basis. Sometimes it's 6 people working a lot, these days it's more like 1-2 who are too busy with their day jobs to do much. Hence why the next major release is more than 6 months "late" compared to original expectations, and there is no indication of when it'll be out.

vinito, 2019-01-06:
Understandable. Thanks for the reply.

Ozzy33, 2019-06-18:
> Cut the drama, **it has been said it would be made configurable**.
> 
> It also has been said that **it requires a storage update**, which means **it needs to wait for the next major release**, which means **dont expect it before some time mid next year**.

I updated a X7 from otx v2.2.0 to v2.2.3 (companion was not displaying output channel monitor movement in the simulator), after the update, it now works as expected. But now, you have to hold the X7 power switch to turn ON the radio, is there a option to make it the way it was before, a single push to turn it on? Maybe it's in there already (name of it?) or do i still need to wait for upcoming revision? tia

PR #6629 (2019-08-17): Add support for pwr-on and power-off speed
closes #4639 

TODOs:
- [x] radio setup for other screen sizes.
- [x] translation strings for other languages (only english for now).
- [x] companion support.
closed 2019-08-17 by bsongis