opentx/opentx #4640: Improve Taranis Latency

Ozzy33, 2017-03-19:
It has been said the Spektrum system has less latency in many threads and on TY.
https://www.youtube.com/watch?v=_rxAPlYr9FQ

There is a 0-9ms latency inconsistency due to the OpenTx OS is not synced to the XJT heartbeat signal, Mike B has done this Tx OS to RF module syncing on erSky9x, should be capable doable in OpenTx.

I found this posts by Mike Blandford:
---------------------------------------
I've done a very specific latency test using a Taranis plus (internal XJT) and a X8R (both using non-EU firmware). This is a test of the FrSky latency leaving out any latency in openTx/ersky9x. The protocol was set to PXX/D16.

I created a very specific pulse sequence on channel 6, and generated a 'scope signal as the PXX frame is created just before it is sent to the XJT. I then triggered the 'scope from the change in the output pulse from the X8R.

I noticed the X8R output pulses were synchronised to the XJT heartbeat pulse.

As the 9mS timing on the radio is just very slightly different to that of the XJT, the time between the PXX signal and the X8R output pulse slowly reduced to a minimum of about 10.4mS, at which point it jumped out to 28.4mS and proceeded to slowly reduce back down to 10.4mS.

Every 9mS, the XJT sends 8 channels, alternately channels 1-8, then 9-16, so channels 1-8 are only sent every 18mS, which explains the jump of 18mS from 10.4mS to 28.4mS.

By synchronising to the heartbeat signal, it should be possible to fix the latency to either about 11mS or 20mS, depending on the phasing between channels 1-8 and channels 9-16.

Since it takes around 3.5 mS to send a complete PXX frame, it would be possible to send all 16 channels within a 9mS frame time. Whether this would reduce the latency is subject to test.

My investigations continue!
Mike.

Edit: I added something to synchronise to the heartbeat signal. The result is the latency fixed at either 11.2mS or 20.2mS.
I did try sending the PXX data twice as often, but the result is the Rx lost signal at times.

ANOTHER POST:
PXX sends 8 channels every 9mS to the XJT.
The XJT also sends a "heartbeat" signal. I use this in ersky9x to synchronise the PXX data to the XJT to minimise latency, otherwise you can get a 'random' extra latency of 0 to 9mS. I'm not sure that openTx uses this signal.﻿

RealTadango, 2017-03-21:
Sending twice as much should not be a problem. FrTX is sending MUCH more frames to the XJT module and it works fine.

And how to deal with dual XJT modules? Is there a heartbeat from the second module also even with telemetry fully disabled?

kilrah, 2017-03-21:
If we do something it would be nice to do it right, since we already already schedule calculations based on when we transmit it would be cleaner and more optimal to align that to heartbeat than to spam the module with less recently updated data or increase CPU consumption with more frequent calculations than needed. 

But there's care to be taken for spurious heartbeat signals as in the dual XJT case. I don't know if Mike has hardened his implementation for this case, but if not done properly receiving a heartbeat pulse while processing the previous could cause problems.

On the other hand, most of this latency nonsense is only there as a marketing con and I think we've got much more important / interesting / actually useful stuff to do than play that game and by that help "justify" all that "omg you absolutely NEED to shave off those couple of ms! Get our new XXX stuff!" BS.

RealTadango, 2017-03-21:
I am very interested in a total latency test from stick input to FC reception. If that is close to the latency differences it makes sense to optimize it.

For planes nobody cares... :) 150msec added delay is not noticeable by an experienced flier....

projectkk2glider, 2017-03-21:
And why do you think that it matters when using the FC!? 

What is important in such case it the **local loop latency** - ie accelerometer - CPU - servo outputs - ESC - motor response time is what it matters for fast multicopters and not so much the radio link latency.

kilrah, 2017-03-21:
That. Many people don't understand that and say "we've now got FCs that look at xx kHz, the radio link has to do the same!" but no. It's still a mechanical thing, and while local loop speed certainly improves stability and response (up to a point) the response to controls still has inertia to overcome.

RealTadango, 2017-03-21:
True but the loops are mainly 1khz or better so 1msec between calculations. This is not true latency but it is what counts for most, and i can agree. With all esc / motor optimizations latency in the quad is getting lower and lower.

projectkk2glider, 2017-03-21:
Another popular and similar marketing BS is the stick ADC resolution.

Ozzy33, 2017-03-21:
If Taranis is in 9ms mode (D16 1-8ch), but its really 20~24ms, heck yes it matters (well not to most of the plane guys). Going from D4R to X4R was night and day, next was 1-16ch to 1-8ch, another noticeable improvement (for good quad pilots), why not drop 1-9ms off, it's not a few ms, its 50% less. It's already been done on ersky9x, why not do it to OpenTx?

RealTadango, 2017-03-21:
No, it is not 50% less of total probably, so wee need to know total first. But it could be noticable...

kilrah, 2017-03-21:
I just had a look at some blackbox logs... the time between the quad receives an order to move and the time it actually **starts to move** (motor response is near instant thanks to fast loop, BUT the mass has inertia and takes time to respond as mentioned previously) is 70ms. 9ms (at best! average gain would only be half of that!) is negligible in comparison. 

> Going from D4R to X4R was night and day

But from D8 (with PPM I guess) to D16 (SBUS) you shaved off about 10 times more than those average 4.5ms.

> It's already been done on ersky9x, why not do it to OpenTx?

Go ahead and do it, no prob! 
But unless someone shows a blackbox log of their quad mechanically responding in <25ms to an input (mine is certainly not tuned like the pros) I'll still consider our developers' time is best spent elsewhere.

Ozzy33, 2017-03-21:
"Many people don't understand that and say "we've now got FCs that look at xx kHz, the radio link has to do the same!" but no. It's still a mechanical thing, and while local loop speed certainly improves stability and response (up to a point) the response to controls still has inertia to overcome and that already represents more than RC link latency."

That's not true at all, I can move the stick from center to full in 40ms, but it only takes a few ms to move it a little (like corrections), say 5 or 10ms, then the sampling of the stick to output of the Rx is ~24ms, 5 times longer than my stick movement took. I am now over 50 and I can feel the difference in lower latency.

"On the other hand, most of this latency nonsense is only there as a marketing con and I think we've got much more important / interesting / actually useful stuff to do than play that game and by that help "justify" all that "omg you absolutely NEED to shave off those couple of ms! Get our new XXX stuff!" BS."

Have you tried PPM and SBUS, can you tell a difference? By staying BS, clearly you're not in the high preformance quad racing and that is fine, this stuff does make a difference to us. Why did Spek spend a lot of time to minimize latency from stick to Rx output on the first 4 channels, because top pilots can tell a difference. We don't need it on all the channels or the "second" ext module. We drop the X4R from 16 to 8ch to get a noticable improvement. The new racing quads are pushing the limits of control, a lot of us laughed when they said we can do a flight control looptime of 1KHz, but it turns out it was much better than 300Hz, then 4K, 8K, 16K and now 32K, every one has made an improvement. Happy Flying!

Ozzy33, 2017-03-21:
My quad from hover to 1400d/p/s is ~60ms, most racers use about half that roll rate, and corrections are a part of that, so under 20ms.

Ozzy33, 2017-03-21:
_"It's still a mechanical thing, and while local loop speed certainly improves stability and response (up to a point) the response to controls still has inertia to overcome and that already represents more than RC link latency."_

Apparently not, Taranis 20+ms, quad under 10ms.

_"People's capability to anticipate is 10 times more important than latency. Problem is BS marketing is playing the "get the fastest gear so you can do like the pros!" which is completely incorrect. The pro can do well with 3 year old gear, "_

I ran the D4R for a couple years, it was fine, as I did not know any better, hell I raced rc onroad cars for 2 decades, with the fastest Airtronics servos (60ms), that's what we had, is it better now with lower latency RF link, you bet. Anticipation is only part of flying/racing, then after you anticipate and make a move, then the small corrections are needed and lower latency makes it feel more connected to the vehicle. On the same subject, I tried RC nitro cars, it felt like a rubber band between your throttle finger and output power, where electric was much more 1 to 1 connected. Have fun!

MikeBland, 2017-03-28:
I don't think my code needs any changing if you have two XJT modules. You can't sync. to both and as the heartbeat signal is open collector, wire-ored, you can't pick out a specific heartbeat.
All my code does is adjust the period, for sending PXX data, if the heartbeat signal is more, or less, than 4mS away from when we run the mixer and send the data.
The period is never more than 10uS away from 9mS. This means you might take around 500, 9mS periods to get into sync, but that is less than 5 seconds.
If you have two heartbeat signals, then the period will just stay 10uS away from 9mS and you won't get sync.

Mike.

schwabe, 2017-03-28:
Can you point at your implementation? I will take a look how feasible it is to port this to opentx.

MikeBland, 2017-03-28:
Ersky9x sources are here: https://github.com/MikeBland/mbtx
Look in radio/ersky9x/src
The code is in drivers.cpp (drivers.h) and pulses_driver.cpp.

In pulses_driver look at TIM1_CC_IRQHandler.
In drivers look at:
struct t_XjtHeartbeatCapture XjtHeartbeatCapture ;
void init_xjt_heartbeat()
void stop_xjt_heartbeat()
EXTI9_5_IRQHandler()
and also check for calls to init_xjt_heartbeat() and stop_xjt_heartbeat() when the heartbeat input may be used for another purpose (trainer input).

The EXTI9_5_IRQHandler() must run at the highest priority so it has no latency itself in reading the timer.
On ersky9x, I've very carefully allocated interrupt priorities so this is the only interrupt at level 0.

Mike.

Ozzy33, 2017-04-03:
This guy (GadgetMart) pretty much hates the Taranis, I quote him in the rcg Taranis thread: "it is supposed to have a low latency rf module. Up to 60% faster. Like the Horus."
Found here:https://www.rcgroups.com/forums/showpost.php?p=37231914&postcount=51221
He is talking about the Taranis with CF paint and M9 gimbals shipped, is there any truth to this or is the heartbeat now working or? 
**Schwabe**, any update to you looking into the heartbeat sync?
Thanks Mike for posting your info!
TIA!

schwabe, 2017-04-03:
@Ozzy33 planing to, when time permits, but that might not be anytime soon. So if you really need the feature now, don't bet on me.

That low latency rf module is probably nothing substantiated. I am not aware of any differences in terms of latency in the Taranis/Horus modules. And all Taranis (X9D, X9D+, X7D) modules run the **same** firmware.

Ozzy33, 2017-04-03:
Thanks for the reply, I am not looking for anything real soon, thanks for looking into this.
You mention the FW is the same in the RF modules, but "I think" the improved low latency comes from  improved OpenTx FW side (syncing with the modules heartbeat signal).

kilrah, 2017-04-04:
This guy has been on my ignore list for quite a while, he's just a troll. Most of his posts are intentional provocation or unreasonable comments often backed with wrong "info".

kilrah, 2017-04-04:
BTW, looks like your data was missed by some of us (you edited your posts instead of adding new ones, so no notification...). It does show that the improvement may be worthwile indeed and was what we wanted before considering time spent on that to be an interesting investment.

Ozzy33, 2017-04-04:
Thanks! Yea, I am a total noob at github, I am not even sure how to correctly quote someone's post, doh! Thanks again.

mythreeleggedcat, 2017-04-06:
I have been following this idea for a while and here are a few comments and notes in case it is helpful to whoever tries to implement this...

The Horus IXJT is different from the XJT and FrSky have stated that it has lower latency and probably is "faster". Apparently FrSkyOS sends updates to the IXJT more frequently than every 9ms and that may account for their claim to have lower latency on the Horus.  They don't need to sync if the FrSky OS sends PXX data a lot faster than 9ms - say 2x 8ch PXX packets per 9ms frame -  because the IXJT will always have the latest data when it sends..either lo8 or hi8.  I suspect OpenTX 2.2 sends PXX packets every 9ms on the Horus and so gets more or less the same varying latency as the X9D+.  

According to Mike B's findings, the X9D+ XJT (Q-X7 is likely the same) apparently cannot tolerate PXX updates twice as  fast and it dropped transmissions when he tried.  Sync is a good solution in that case.  One issue with it is that the heatbeat is every 9ms and does not indicate if the XJT transmitted lo8 or hi8.  To get the best latency you have to restrict the PXX interface to sending only 8 channels.  Then the XJT will either optimize to sending lo8 every 9ms (it is widely believed it does) or if it does not optimize then at least the lo8 data will be fresh each time the lo8 packet is sent every 18ms.  If you send 16 channels in 2 PXX alternating packets, one every 9ms, then you risk being out of phase and then the latest data has to consistently wait > 9ms before the XJT will send it.

From what I can gather the D16 protocol does allow optimization to just sent the lo8.  I haven't seen anything definitive that indicates it actually does this although there are plenty of anecdotes.  I'm pretty sure that the channels are ordered 1-8 and 9-16 in an array but not named.  I think there is a flag or something that says if the array is lo8 or hi8.  

Another point is that the external module heatbeat pin is apparently used by the "Linker" wireless trainer feature so presumably sync'ing the Module and using the "Linker" wireless trainer feature will need to be mutually exclusive either via a build option or via some other configuration.  Presumably Mike B has handled this in er9X so the same solution might work for OpenTX.  
http://alofthobbies.com/linker.html

The Horus IXJT cannot run exactly the same firmware as the X9D XJT/ Q-X7 XJT/ XJT because FrSky made fixes to it on the Horus to i) allow both antennas to be used for TX diversity and ii) to fix some kind of lockout that was causing RX failsafes.  The original XJT does not have an antenna switch and as far as I am aware has not had that lockout problem.

GadgetMart is likely mistaken in suggesting that the X9D+ has an IXJT.  I have not seen that stated anywhere else.  Without OpenTX being changed to send the PXX packets either in sync or twice as fast there probably no significant improvement so why would FrSky go to the expense and trouble?  Also technically they may have to re-certify with the FCC if they put a new TX module in the Taranis case.  Everything else on the SE is just mechanical assembly stuff.  

Mike's original post...with note about the heatbeat pin and Linker
https://www.rcgroups.com/forums/showpost.php?p=34821735&postcount=48226

Mike's findings...
https://www.rcgroups.com/forums/showpost.php?p=34823968&postcount=48240

A discussion of the phasing issue
https://www.rcgroups.com/forums/showpost.php?p=34827101&postcount=48265

RealTadango, 2017-04-06:
FrOS updates the external XJT much faster also then OpenTX ;) The module can handle it :)

kilrah, 2017-04-06:
That's exactly what he said above...

The fact that nobody knows *exactly* what the XJT does is sure also putting off from making much effort there. Many things would need to be tried and mearurement tools developerd to really be sure the right thing is done.

3djc, 2017-04-06:
And also at least one other device is affected,  but how many others out there would be too ? There are A LOT of OpenTx users, with extremely varying needs and usage, that's need to be factored in too.

schwabe, 2017-04-23:
@MikeBland I started looking into implementing this on the Taranis. Since we already have a dedicated timer for the heartbeat/Trainer pin. Is there reason not to start this timer from the xjt output timer (with the stm32 timer sync support) and use the already existing code that captures trainer inputs to parse the heartbeat signal instead of the highest priority interrupt? Or am I overlooking something.

MikeBland, 2017-04-23:
As long as the timer is not doing anything else then that should work. In my case, I have a single timer (timer 3) that handles trainer input and output on the trainer jack socket, and can also handle PPM trainer input on the heartbeat pin.
I don't see any need to synchronise the timer to the XJT output since the position is calculated using:
XjtHbeatOffset = TIM7->CNT - XjtHeartbeatCapture.value ;
so the current timer value is compared to the captured value.

mosh1, 2017-04-30:
As a long time flyer of high performance aircraft (3D Helis, 100+mph jets, racing quads, etc) I'm really happy to find effort into this since it's been one of my biggest complaints with my Taranis when compared to my Spektrum systems - the improvement in 'connected' feel is absolutely better the lower input latency there is. A question - do we know yet if efforts here improve latency for external modules such as the TBS Crossfire module (using the CSRF protocol to module)? I would definitely go the extra effort in configuring special build options if need be as mentioned previously.

kilrah, 2017-04-30:
CRSF is already "ideal".

mosh1, 2017-04-30:
@kilrah interesting, so right now end to end latency from stick to FC input is faster on Taranis with CRSF than with Taranis internal module and FrSky rx?

kilrah, 2017-04-30:
No idea, that depends whether the crossfire system is faster than the FrSky one or not :)

All we know is that OpenTX->module communication has no known room for improvement for the CRSF protocol, while there is some for FrSky modules. Whatever happens afterwards is kind of out of our scope.

schwabe, 2017-04-30:
As a side, I am on implementing this (but takes time, etc.). I will probably first implement this feature for the [multprotocol](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module) as I can control both sides there.

FieserKiller, 2017-08-23:
Hi, 
are there any updates on this topic?

Matthias7Song, 2017-08-23:
Hi, are there any updates? I'd want to know if I have OpenTx2.2.0 on Taranis and set D16 1-6ch(9ms), then the receiver refresh rate should be 9ms and no more latency.  I have looked into OpenTx 2.2.0 RC release notes and 2.2.0 release note but have not yet noticed the latency issue.

schwabe, 2017-08-23:
no, no updates on the point. I have working prototype code but I pretty much abandoned it for the internal XJT since I have no good solution to determine the "good" value for the difference between heartbeat and our output. @MikeBland experimentally determined a good value on his X9D/XJT firmware combination but I am not sure how universal that value is. The timing might be completely different for other FW revisions or other IXJT modules (X7, Horus...)

schwabe, 2017-11-13:
For people who want to check out input synchronisation, the multimodule code on my multi repo (https://github.com/schwabe/DIY-Multiprotocol-TX-Module) and current OpenTX has code to minimise the delay. 

I will not port the code to XJT/PXX modules unless Frsky add some telemetry or documents the heartbeat to signal output relation.

Ozzy33, 2018-01-08:
schwabe, PM me your email address in rcg, I have some news for you, thanks.
https://www.rcgroups.com/forums/member.php?u=91160

schwabe, 2018-01-08:
@Ozzy33 my email address is pubically visible on github anyway: arne@rfc2549.org just use that.

fredericmorin, 2018-02-26:
@Ozzy33 why not posting here to keep everyone up to date?

Ozzy33, 2018-02-26:
Convoluted info, will update when something good comes around.

fredericmorin, 2018-04-12:
@schwabe @Ozzy33 any news ?

RCdiy, 2018-04-21:
This video has some interesting results regarding latency.
https://youtu.be/75E1WSrwqaY
He compares flicking a switch and the time it updates on the output from the Rx. Many frames get skipped (as high as 5) before and update propagates.

MikeBland, 2018-04-25:
That may not be a true representation of FrSky performance. I've just repeated some bench testing using a QX7 (with ersky9x) and a X8R, both on EU (LBT) firmware.
I first compared the SBUS output to the channel 1 servo output. Both start at the same time and every time the servo pulse changed, the SBUS output had the corresponding output. Since the channel pulse occurs every 18mS and the SBUS is sent every 9mS, it is possible the SBUS output did change 9mS earlier, but I didn't check that.
I was working in D16 mode with 16 channels.
I then looked at changing a switch to control a channel output. I obtained a minimum delay of 16mS and a maximum of 52mS (36mS later). Working on the bench however, I noticed the green LED on the Rx was flickering, suggesting the Rx was swamping, and so missing some RF frames. If a frame was missed, this would add 18mS to the delay. I suspect the same is likely true for the testing done in the video(s).
(You pointed me to https://youtu.be/P3OI6LMtF6c for FrSky testing).
The 52mS delay only happened occasionally, mostly the delay was much shorter.
Note that ersky9x is using synchronisation between the XJT heartbeat signal and the transmission of a PXX frame to the XJT.
Both openTx and ersky9x may also be sending the failsafe data occasionally. When these PXX frames are sent, then there is an extra 18mS delay in channel update data. It is possible this also occurs in the testing shown in the video.
As usual, some results are shown, but without proper data collection. He takes the minimum and maximum delays, then presents the results as the mean of them, without checking that the actual distribution of delay values between them is constant. The actual "average" delay may well be shorter.
I'll do some more testing when my Tx battery has charged!

Mike

RealTadango, 2018-04-25:
If you disable failsafe, will that eliminate the failsafe frames?

RCdiy, 2018-04-25:
He provides his spreadsheets.  Uses 7-10 samples.

MikeBland, 2018-04-25:
A bit more testing. I built a special version of ersky9x that outputs a signals on the two "spare" switch inputs, then used one of these switches to control channel 1. This makes it easy to 'scope the "switch", and also provides a 'scope loop to make seeing what is happening easier. The signals were set to change state every 50mS.
The results are a bit inconclusive! I tried a X8R, and a brand new S8R, although the S8R doesn't have recent firmware as it doesn't have "Quick mode".
I triggered the 'scope on the channel pulse from the Rx changing from 1 to 2mS, then looked to see when the switch input changed. With the X8R, the shortest delay I saw was 36mS, with the actual delay changing continuously. With the S8R, the shortest delay was about 37mS, but stayed at that delay most of the time.
What I also observed is that over time the delay slowly drifted. no doubt due to the slight difference in clock speed between the radio and the XJT. Interestingly, I also noticed that if I switched the Rx off, then back on, the delay started with a significantly different time.
I suspect that different firmware in the XJT and the Rx may produce different results, so I probably need to update everything to the "latest" before I do any more tests.

Mike

kilrah, 2018-04-26:
Is that drift using heartbeat sync or not?

MikeBland, 2018-04-26:
Yes, the drift is still using heartbeat sync!
The automatic switching of the switch is not in sync with the heartbeat. I don't know if the Rx pulse output is in sync with the received RF data. I'll try to check that by monitoring the supply to the Rx. I can usually detect a small dip when the Rx sends a telemetry packet.

Mike

kilrah, 2018-04-26:
For a "proper" evaluation one should also decode the sbus stream instead of looking at the servo pulses. Nobody who cares about latency nowadays uses PWM or PPM...

MikeBland, 2018-04-26:
Which is why I checked that first, as I posted earlier:
"I first compared the SBUS output to the channel 1 servo output. Both start at the same time and every time the servo pulse changed, the SBUS output had the corresponding output. Since the channel pulse occurs every 18mS and the SBUS is sent every 9mS, it is possible the SBUS output did change 9mS earlier, but I didn't check that."

Mike

MikeBland, 2018-05-02:
Some further testing. I've updated my Arduino code for SBUS to PWM output so it detects the SBUS frame where channels 1-8 are updated (16 channel mode from Tx) and synchronises the output pulse generation to that for minimum latency.
I then compared the channel 1 output from the Rx and the SBUS decoder. Sometimes the Rx pulse appeared first and sometimes the SBUS one appeared first, but the order remained fixed until power off. It appears the Rx chooses an 18mS period for the Rx outputs, and sticks with it without attempting to synchronise with the RF data, so you get a random extra 9mS latency to the Rx outputs.

Ozzy33, 2018-05-02:
FYI, I have a very nice $17K Tek DPO4104B 1GHz 4ch scope with all the probes (current, diff, etc) at work I use, but it's a little bit of a hassle zooming in on each packet (even using the wave inspector search function). So I just ordered one of these for $9 https://www.amazon.com/Alloet-Logic-Channel-Analyzer-Debugger/dp/B01MR0EV87/ref=sr_1_8?ie=UTF8&qid=1525147361&sr=8-8&keywords=USB+Logic+Analyzer and get the software here https://www.saleae.com/downloads

MikeBland, 2018-05-02:
I have a 2-channel Textronix digital storage 'scope (1Gs/s), but I also have a similar logic analyser I use. I did, in fact, just use the logic analyser to confirm the extra 9mS latency you might get. My SBUS decoder was synchronising to the SBUS frame, but the Rx output sometimes was synchronised and sometimes was 9mS later, depending on the time of Rx power on.
The firmware I have on the XJT (in a QX7) and the X8R is the "latest" EU LBT, including the bind options to select Rx channels and telemetry.

mythreeleggedcat, 2018-05-06:
Tandango wrote "If you disable failsafe, will that eliminate the failsafe frames?"

Depends what you mean but it is unlikely to matter.

There is not really such a thing as "disabling failsafe".  You can just set the failsafe mode to one of "Hold"  (the default), "Position", "No Pulses" or "Reciever".  "Not Set"  essentially means "Receiver" and (as far as I can tell) just provides a way for OpenTX 2.2 to give you a warning you haven't set anything explicitly.

According to the D16 analysis done by the MULTIPROTOCOL folks, the failsafe packets work by sending sequence to 16 packets (or maybe just 8 if only 8 channels are being used) approximately every 9 seconds.  These failsafe packets are a variant of the regular control packet.  Each packet in the sequence updates the failsafe setting for one channel.  The channel's failsafe mode is sent and if the mode is "Position" then the channel value contains the channel's failsafe position.  In that packet, all the other channel position values are sent as normal active control values. Probably the channel value is a normal active control value if the mode is not "Position"

Assuming that analysis is correct,  this will have a completely negligible effect on latency because in the worst case when Position is use,  each channel only has to hold for one packet every 9 seconds while the other channels operate as normal.  With a packet every 9ms that's a 1 packet channel hold once in a 1000 packets.  Further,  in general you only need to set "Position" failsafe on the throttle and active control surface channels and when using a Flight Controller, you generally only need to set positional failsafe on the throttle channel, a spare channel or possibly none if you use SBUS which has a separate FS status flag.  Assuming Failsafe packets send a control value, unless "Position" is being set then in the SBUS case there are no missed packets caused by this feature.

Since the Failsafe mode can be set per channel and could be changed at any time then it seems likely the D16 failsafe packets are always sent by the XJT.  I do not know how OpenTX controls this.  Possibly it sends the Failsafe settings when the XJT is first powered on and again when a change is made.  In this case the XJT must remember the setting and creates the packets on its own ~9s schedule.  Possibly the XJT updates the XJT regularly and the ~9 second timing comes from that update.  I'm sure those here familiar with PXX can say.

MikeBland, 2018-05-13:
8 channels of failsafe are sent in a single PXX packet. The 9S timing comes from the radio firmware sending the failsafe PXX packets every 1000 packets (9mS per packet).
With ersky9x, I now have a latency from input to SBUS output of between 14.4 mS and 23.3 mS.
After talking to FrSky, I understand it is permissible to send 2 PXX packets in a 9mS period. This is what I am now doing (actually it is an option), sending channels 9-16 first, then channels 1-8 just before the heartbeat pulse. From a logic analyser capture I can see this working (I'm capturing the SPI internal to the XJT to the CC2500). Doing this has reduced the maximum latency by 9mS.
The Rx SBUS output does change as soon as possible after receiving a RF frame with a new value in it, so may change 9mS before a servo channel output.

MikeBland, 2018-05-15:
Using "double rate" PXX (two frames in 9 mS), and sending 16 channels, or "single rate" but only sending 8 channels, with synchronisation to the heartbeat signal) I obtain the following latency values in ersky9x.
Servo pulse and SBUS.
Minimum time: 14.4 mS.
Maximum time: Servo pulse 32.3 mS, SBUS 23.3 mS.

I've put openTx on the Taranis plus (release 2.2.1) and repeated the latency measurements. Because openTx doesn't (currently) synchronise to the XJT heartbeat pulse, and, in 16 channel mode, only sends one frame to the XJT every 9 mS, there is quite a variation in the latency, slowly changing over time.

The shortest latency I found was 19.3 mS, jumping to 37.3 mS, and the longest I found was 53.9 mS jumping down to 36.1 mS. These are times from control input to servo pulse. With the longer times, sometimes the SBUS output does occur 9mS before the servo pulse, so the 53.9 mS was 44.9 mS to the SBUS output.

Ozzy33, 2018-05-15:
Hi Mike, Thanks for all your help on this stalled project. 
These numbers are truly embarrassing for Frsky. Most pilots don't notice for a few reasons.
1. Planes have a huge reaction latency do to many factors adding up. (I quite flying my planes do to this)
2. Pilots get used to the latency and adjust to it.
3. Not all pilots can feel the difference in lower latency.
4. Type of flying.

A decent quad on the other hand can start to response from Rx output to start of roll/pitch in as little as 3ms. Then add on 54ms for the Tx processing and Rf link, lol.

I have got stonewalled from Frsky and AirPixel (frsky warranty and sales) for any help.
Thanks again Mike!

MikeBland, 2018-05-15:
Be aware the actual latency varied over time. Due to sending channels 1-8 only once every 18mS (16 channel mode) The latency from an input at any one time has a minimum and a maximum that are about 18mS different.
The actual latency drifted over time because the PXX data is not (currently) synchronised to the XJT heartbeat.
I saw:
25.6 and 43.4
21.3 and 39.1
19.3 and 37.3
30.2 and 48.0
34.8 and 52.6
36.1 and 53.9
and with the maximum times, the SBUS output was usually 9mS earlier.
At the moment, I can't explain why the latency is as long as 53.9mS at times, but from my measurements, it seems much of this time is within openTx due to phasing between the PXX frames and the RF frames, as well as phasing between the input and the mixer running.
I'm fairly certain that if the "double rate" PXX is used, and the PXX is synchronised to the heartbeat, then the latency will be similar to that I have in ersky9x (14.4 mS and 32.3 mS [23.3 mS to SBUS]).

kilrah, 2018-05-16:
> as well as phasing between the input and the mixer running.

We always sample inputs 2ms before sending a PXX frame so that should be constant and low.

[187 more replies]

closed 2019-06-12 by 3djc