opentx/opentx #4641: Multi subtype broken on ar9x

kilrah, 2017-03-19:
Subtype name not even appearing

schwabe, 2017-03-20:
It works fine on my AR9X simulator and current next: 

[image]

is that just on the radio itself?

kilrah, 2017-03-20:
Possibly then, someone reported it dodn't work on their radio, I checked and it doesn't work on mine either

[image]

schwabe, 2017-03-21:
that might be because of the way how I store the strings. Are String on Atmels arm somehow special (like AVR's strings)?

kilrah, 2017-03-21:
Yep I do have the 2 0's in Custom mode!

bsongis, 2017-03-21:
I don't think this commit will fix the issue though!

kilrah, 2017-03-21:
It did!

closed 2017-03-21 by kilrah

bsongis, 2017-03-21:
cool then :)

schwabe, 2017-03-21:
I don't really understand why that commit fixes it. But at least it is fixed now.

PR #4656 (2017-03-21): Define multi subtype strings in the same way as all other strings
The removal of the static might fix #4641 if there is some linking problem. Anyway even if it does not fix the bug, the increased consistency is nice.