opentx/opentx #4659: RC13 N365: LUA panic / not enough memory with BF lua script

RipperDrone, 2017-03-21:
RC13 throws memory error when installing BF lua script: 

https://github.com/betaflight/betaflight-tx-lua-scripts/tree/c51d80d24e7a2f53f0bfc72d1d64c48a06031514

Will ask the BF dev team for analysis, too...

kilrah, 2017-03-21:
Try both versions on RC12, if you've got the same result it's not on our side.

RipperDrone, 2017-03-22:
Hmm, radio shows some 19k free memory, doesn't indicate low memory actually...

RipperDrone, 2017-03-22:
@kilrah has been fixed:

- I had non-lua firmware version installed (for some reason, BF lua scripts have been working without the lua / luac check boxes ticked throughout v0.3, though).
- PID values were lost due to a bug in the script. Has been fixed now.

closed 2017-03-22 by RipperDrone

kilrah, 2017-03-22:
Standalone and telemetry scripts don't need lua ticked.

RipperDrone, 2017-03-22:
I see, this explains it 👍 Thank you so much.

kilrah, 2017-03-22:
Well it precisely doesn't!

RipperDrone, 2017-03-22:
rev 0.3 of the script was a standalone version. Splitting it into multiple files came post v0.3, so this might have rendered the script into "non-standalone"?

kilrah, 2017-03-22:
No. The option refers only to mixer scripts.

RipperDrone, 2017-03-22:
ok - well, then you are leaving me partly puzzled. Still happy it's working now...