opentx/opentx #4661: LUA Widget options.  Dealing with persistent options when script options change.

agibson2, 2017-03-21:
Widgets save options to persistent storage for an instance of a Widget.  This allows Widget options to be persistent across power on/ power off of the transmitter.

If the options changed from one version of the Widget to the next version, the saved options will not be in sync with the new widget code. I wonder how that can be dealt with for the user. For instance... version one has 3 options. Version 2 has 4 options. Or maybe option 1 was changed from SOURCE to VALUE. When they update the widget files on their transmitter to the next version of the widget, the saved options will not be in sync with the new LUA Widget code.

Maybe some kind of CRC on the option names and their type stored with the options so that the transmitter will know to reinitialize them again on power on if they are different or is this handled already somehow?  If OpenTX has the actual name and type saved then maybe it could only invalidate the options that changed types or new ones that were added.  Maybe also removing any that are not any longer in the script.

3djc, 2017-03-22:
Well i think that like all developers, the dev of the widget need to implement the handling of that. They can add versions to their widget name for example. It would be hard to devise a solution that always works  nicely without knowing what the widget does and the changes between versions

mpaperno, 2017-03-22:
One way to handle such things is to save a version number along with the saved widget state (options).  Then one can know which data to expect when reading the back the saved data.  e.g.  https://github.com/opentx/opentx/blob/next/companion/src/simulation/simulator.h#L143

closed 2020-04-06 by stale[bot]

agibson2, 2020-04-07:
This is still a problem from what I understand and should be looked at still for 2.3.

lshems, 2020-04-08:
It is not a problem in a sense that a new VERSION is not a concept existing for widgets. As long as thereis no proper version mechanism, this is not possible to implement.

My 2 cts.