opentx/opentx #4663: RSSI and Rx Battery voltage not actual values

mpjf01, 2017-03-22:
At present it seems that if telemetry is lost the RSSI and RxBatt telemetry values being displayed on the screen and being logged are maintained at whatever they were when the last valid value was received. (If the radio is in RT mode the RSSI value shown in the RT window is the actual value (zero) in this situation.)

Clearly there are reasons for not displaying and logging the actual values being received as OpenTx was changed from displaying actual values to displaying last good values. 

There are also reasons why it is desirable to be displaying and logging the actual values.

Would it be possible to give the user the option over which situation was to apply? Maybe as a different compile option or a setup option in the radio or model setups.

RealTadango, 2017-03-22:
Same issue as https://github.com/opentx/opentx/issues/4629

RealTadango, 2017-03-22:
#4629

3djc, 2017-03-22:
I guess it can be debated if 0 is a more actual value than last one, since there is no new value. The fact that you do not received telem anymore doesn't mean the receiver is receiving 0 volts, or the rssi is 0 (actually, you might very well  still be in control of the craft).
Don't think a per model or compile option would work very well with scripts.
Maybe we can have more info on the situation where you find it been an issue ?

kilrah, 2017-03-22:
I believe the display shouldn't be changed (it already should signal the lack of data either by flashing the numbers or displaying NO DATA), but it would be good to have a "No data" or "invalid" indicator in the logs, which could be 0.

3djc, 2017-03-22:
or not logging it at all ?

RealTadango, 2017-03-22:
Mmh, making the value missing is a very common solution.... the graphing tools could show them by a different colors / dotted line to indiate no data was receivers for that sensor value.

kilrah, 2017-03-22:
> or not logging it at all ?

Yes why not, might need siginificant changes in companion though, need to also make sure that works with other way to graph (e.g. excel)

RealTadango, 2017-03-22:
After what time do you consider it missing? The same timeout as for the sensor lost warning?

kilrah, 2017-03-22:
no much shorter, about 1 sec I believe.

kajsilander78, 2017-05-07:
Why not add a variable that zeroes out on telemetry lost, that would give the option to reset any telemetry you like with "Reset" under "special functions"?

teckel12, 2017-11-18:
When telemetry is lost, we know that it's lost, and I think it makes the most sense to see the last value.  An individual telemetry sensor can be lost, not all.  So in all cases, it makes sense the way it is now.  We know telemetry is lost, and we get the last good value.

mythreeleggedcat, 2018-07-17:
Since there is now an implicit Tele "switch" in OpenTX 2.2.X  which has states Tele and !Tele,  why not log that along with the other intrinsic switches?  This would explicitly indicate in the log when telemetry is being received.

closed 2019-11-15 by stale[bot]