opentx/opentx #4664: [Horus] Serial port options not implemented / Emergancy mode

RealTadango, 2017-03-22:
The serial port options under the Hardware section is not working / implemented. When set to SBUS trainer the internal serial port header does not accept any S.Bus input. When set to Telemetry the Horus goes into Emergancy mode.

This might be related to https://github.com/opentx/opentx/issues/3954 if the internal UART header RX pin is the same as the heartbeat pin.

3djc, 2017-03-31:
This option will be made unavailable for 2.2.0 release. We will come back to it and make it work at a later point in time

kilrah, 2017-04-25:
Temp fix in 9ff1dab6b83b15e30aafda03b70732c20b91b54d

raphaelcoeffic, 2019-08-14:
@3djc @kilrah has anything been done here? if it is hidden, there is no "bug" anymore. The feature request is still in #3954.

3djc, 2019-08-14:
there is nothing hidden ...

raphaelcoeffic, 2019-08-14:
> there is nothing hidden ...unless I misundersdant what you mean by that

Unless you have `DEBUG` defined, it seems to be hidden when looking at the code. Did I miss something?

3djc, 2019-08-14:
well it has a serail debug purpose when DEBUG is enabled, but when not, it is hidden since it does not 'do' anything, so I guess we can close that if that was the bottom of your question

kilrah, 2019-08-14:
#3954 is different, it's about the heartbeat pin. This is about the serial port. On other radios we have several functions available on the serial port, on Horus it was broken hence this issue. To avoid people having their radios crash when selecting a function the whole thing was hidden, but the correct fix is to fix those functions.

PR #6733 (2019-09-10): Deactivate aux_serial on Horus targets
closes #4664 

Please note that this still allows for using USB serial (CDC) for LUA or debuggging.
closed 2019-09-10 by bsongis