opentx/opentx #4671: Companion 2.2.0N365 Simulator trims adjustment via buttons not spanning full range

elecpower, 2017-03-22:
All trims -25 to +25 via dragging sliders but in most cases -24 to +24 via up/down arrow buttons. Sometimes -25 or +25 if drag slider away from max +/- and use buttons to return to last max +/-.

Tested for Taranis X9D+ on Ubuntu 16.04

kilrah, 2017-03-22:
Where are you looking at the value? Some places may display 24 due to rounding.

elecpower, 2017-03-22:
[image]
Radio sticks window range +/- 25% via sliders or generally +/- 24% via buttons.
Also checked Radio outputs and channel range is always -/+24%

mpaperno, 2017-03-22:
What does it show for outputs on the simulated radio itself?  That's what we really need to match.  

I think the Radio Outputs rounding issue is fixed by a recent commit.

Those trim sliders are a bit of an anomaly since they don't exist on the actual radio (really just there for convenience).  There is another issue open about them as well (not being constrained when trims are disabled).  Not saying that as an excuse, but it's really the trim buttons which simulate the radio hardware properly.

elecpower, 2017-03-22:
As you say rounding but trim 24% vs 25% does affect the simulated radio value.
[image]
[image]

kilrah, 2017-03-22:
24.2 is actually correct and consistent with the radio.

mpaperno, 2017-03-28:
With current `next` code the only rounding issue I see is in the X/Y indicators next to the sticks themselves.  The Radio Outputs display rounds correctly (also for negative values), and the trim sliders set the same values as using the buttons.  However the X/Y values are not rounded the same way as the outputs right now (same issue as we had there).  Also discovered another issue -- those X/Y numbers don't update on trim change until the virtual stick is moved.

However...  I think the bug is that those X/Y values include the trim adjustments at all.  Shouldn't they just show the virtual stick position?  The final channel output is up to the radio, after all, and that's what the radio output screens are for.  Those X/Y values do not take any other radio outputs into account so I'm not sure why trims should be included.

Now maybe it would be useful to show the trim outputs somewhere in the simulator UI... which could be arranged (say, in the Radio Outputs widget, or next to the trim controls).  But I think those X/Y value by the sticks should show where the virtual joystick is actually positioned.

Opinions?

3djc, 2017-03-28:
I think those value should be stick only too. If you look at input screen on the radio, it does not take trims into account either.

kilrah, 2017-03-28:
Indeed, especially given that the trim isn't even necessarily enabled, but this % currently includes it even if not.
Kinda makes no sense anymore with the inputs and their flexibility.

closed 2017-04-02 by bsongis