opentx/opentx #4678: Companion Simulator for X7 on Win10, RC14n366

rdeanchurch, 2017-03-24:
Am I missing a setting somewhere?  
The simulator for X7 has no switches showing.

and in Companion, the model settings has no Switches showing  position for checking on model selection.
Oops, The switch settings for model selection does not sow on the transmitter 
[image]
either

3djc, 2017-03-24:
[image]

The simu reads the actual radio settings (in hardware section); so for some reason you do not have the switches declreed there I would think.

[image]

rdeanchurch, 2017-03-24:
You are correct.  
I don't see how those went away on my computer, but they were gone.
Thank you.

closed 2017-03-24 by rdeanchurch