opentx/opentx #4681: Different channels behaviour in simulator

OpenMik, 2017-03-24:
Channels visualizations have different behaviour in simulator

[image]

3djc, 2017-03-25:
Duplicate : #4619
Display rouinding error that has been fixed

OpenMik, 2017-03-25:
Companion was downloaded by PPA
I'm on Xubuntu 14.04
today it was compiled from source
[image]
but the behaviour is the same
I suppose that I do something wrong, but I don't know what

3djc, 2017-03-25:
Indeed, I get the same results,  it was marked as fixed, but apparently not

projectkk2glider, 2017-03-25:
I bet rounding is not working correct for negative numbers only.

3djc, 2017-03-25:
Yes, I'm on it

PR #4683 (2017-03-25): Use rouding like arm radios for channel outputs
Thix fixes #4681
closed 2017-03-25 by bsongis