opentx/opentx #4685: 366n simulator Problem with RSSI value (trigger LS)

strgaltdel, 2017-03-25:
Hi,

i guess i found a little issue in the 366n sim:

RSSI value handling seems to be a little bit messed up
in case RSSI runs below a given limit, logical switches will not be activated to trigger some SFs.
(Sensor Autodetect returns unit dbm, LS operates using Db, maybe this is the root cause?)

simple test:
- within companion, in a new model, define telemetry value "RSSI" as usual or perform sensor autodetect on TX to get a RSSI value
- define LS x<40 on RSSI
- open telemetry simulator and reduce RSSI below 30, LS doesnt' trigger !

Tested on X9e, TX firmware behaves without failure, seems to be a sim problem

regards 
Udo

[image]

[image]

[image]

kilrah, 2017-03-25:
Does it do this only with RSSI or also other values?

You don't have the "Simulate" tick enabled in that capture BTW

strgaltdel, 2017-03-25:
Tried with an existing model,
same failure on RxBat, LS doesn't trigger in case of low Voltage.

In case you you trigger the LS on a dedicated GVAR value, or an input value, it does !
For me it seems to belong to telemetry values.

Will "dig a little bit deeper" to rule out that there are some older problems due to step by step migration thru several nightlys.
But RSSI (and Rx Voltage) Problem is definitly reproducable on a brand new (empty) model...

kilrah, 2017-03-25:
The "telemetry streaming" mechanism that disables LSs if no telemetry is received might be using an alternate codepath that the telemetry simualtor bypasses.

3djc, 2017-03-25:
On what radio do you get RSSI autodetected as dbm ?

schwabe, 2017-03-25:
Iirc that mechanism is set in the telemetry parsers themselves whenever a message is received (```  telemetryStreaming = TELEMETRY_TIMEOUT10ms``  Not sure if simu emulates that.

kilrah, 2017-03-25:
@3djc in companion the unit is dBm... maybe need to change it

strgaltdel, 2017-03-25:
@3djc  Radio Type is X9e, but i didn't told it correct:
Tx recognize the unit as db, but Companion shows dbm !

strgaltdel, 2017-03-25:
..there seems to be no db option in companion !

strgaltdel, 2017-03-25:
... sorry must leave the playground, wife made a "last call" for shopping tour ))-:

strgaltdel, 2017-03-26:
just tested the misbehavior on a new win10 installation, same result.
so i can rule out that my installation is screwed up due to a long list of installed nightlys.

3djc, 2017-03-26:
then you need to supply a much detailed steps to reproduce, because we cant

kilrah, 2017-03-26:
Can you answer my question from yesterday?

>  Does the correct value appear on the telemetry page in the simulated radio?

strgaltdel, 2017-03-26:
@kilrah 
sorry, my failure
Just tested, RSSI value at tele screen stays fix at initial value "75", independent fromchanging value in telemetry sim

@3djc 
simple test:
within companion, in a new model, define telemetry value "RSSI" as usual (label RSSI, ID F101, Inst 25, unit dBm

define LS x<40 on RSSI

open telemetry simulator and reduce RSSI below 30, LS doesnt' trigger !

attached you will find an example in the otx file
model#2 is very "lean"
simple look at LS1 by reducing RSSI in the telemtry sim..

[sample1.zip](https://github.com/opentx/opentx/files/870701/sample1.zip)

3djc, 2017-03-26:
ok, if instead of creating the sensor in companion, you autodetect it on the radio (or simu), it works

3djc, 2017-03-26:
Your model works fine. ONCE AGAIN, this box NEEDS to be check for 'virtual' telemetry  data to arrive:

[image]

closed 2017-03-26 by 3djc

strgaltdel, 2017-03-26:
ouuch 
sorry for generating this confusion and misusing your time
BIG AND BLOODY DUMMY MISTAKE ON MY SIDE !