opentx/opentx #4686: Companion profile behavior incorrect

kilrah, 2017-03-25:
- When deleting a profile, another one is loaded. The window contents are however still those of the now deleted profile, so exiting with "OK" uverwrites the settigns of the newly selected profile with that of the old.

- When creating a new profile it starts as a copy of the currently selected one (which might have inappropriate settings configured), should start blank / with defaults.

mpaperno, 2017-03-25:
I kinda like the 2nd bug/feature (that fine line again ;) ).  But I expect it now, and select the profile I want to copy first.  Which I agree isn't really intuitive.

3djc, 2017-03-25:
Specialy not intuitive if your copied profile has data you cannot see anymore because you changed radio type!

mpaperno, 2017-03-25:
If the now-hidden data is **not** irrelevant (as I would expect it to be if it's hidden) then yes, that's a potentially big problem.  I hadn't considered that "artifacts" from unused settings could cause issues.   If true then the underlying data conversion/handling should be fixed, I think (not rely solely on UI to handle anomalies).

3djc, 2017-03-25:
Try this : create x9 profile, assign a boot picture, create an horus profile, delete the picture, you can't flash anymore

kilrah, 2017-03-25:
>  If true then the underlying data conversion/handling should be fixed

Thing is, there is none :) Until now the settings were similar enough (identical?) not to care. Options differ and always have behaved in a somewhat funky uncontrolled way when switching from a profile to another.

mpaperno, 2017-03-25:
> Thing is, AFAIK there is none

Oh.  :) 

There are so many eeprom* and firmware* pieces I figured it was buried in there somewhere...  😆  

So ultimately a converter is needed (perhaps) but for now maybe some simpler way to avoid the bad data.  Does anyone know what the actual issue with Horus is (where it goes bad)?

> The window contents are however still those of the now deleted profile

What do you mean by "window contents" here?  `g.id()` or the MDI window(s) or...?

kilrah, 2017-03-25:
> Does anyone know what the actual issue with Horus is (where it goes bad)?

Horus doesn't support splash screen replacement, so the fields are hidden (in addition to being useless and confusing it's problematic since the splash image size makes the window too tall on some screens). But when you make a new profile when the currently selected one has a splash set, it gets copied. IF that image has been deleted flahsing fails as it can't be found. Normally you would change the splash or erase the reference, but since the fields are hidden you can't. 

Could also skip the splash screen replacement for Horus during the flashing procedure, but I'm afraid to dig into that and not populating a non-applicable setting in the first place would be cleaner anyway.

> What do you mean by "window contents" here? g.id() or the MDI window(s) or...?

The settings -> profile window. But I can't seem to reproduce now... :\

mpaperno, 2017-03-25:
> IF that image has been deleted flahsing fails as it can't be found.

Ah, this is all simpler than I was imagining.  I thought it was some underlying radio data structure error.  But I see how it brings up the bigger issue.

PR #4714 (2017-03-31): Instead of copying current profile, start from a blank one
Thix fix #4686 

Thix changes current behaviour of copying current profile when creating a new one. Given the differences in radio capabilities and working principles, this protects from carrying not relevant, sometime incompatible data
closed 2017-03-31 by bsongis