opentx/opentx #4696: Companion (N366) does not keep Vario source when I import models from Horus

Red-Devil74, 2017-03-26:
When I Import the models from the Horus with Companion (N366, Mac OS X), The Telemetry tab shows three dashes at "Vario Source" while it is set as Vspd (and working) in the Horus.

[image]

3djc, 2017-03-26:
When you simulate this model, is it ok on the simulated radio?

Red-Devil74, 2017-03-27:
That is funny. It shows in the sim but if I enable th telemetry vario, I do not hear the sounds when I raise or lower the altitude. I do not know if it ever worked (like this) in the sim, however...
[image]

PR #4700 (2017-03-27): Companion : fix vario for Horus
This fixes #4696
closed 2017-03-27 by bsongis