opentx/opentx #4702: Missing FR translation on the Horus

LapinFou, 2017-03-27:
Hi folks,

When using Companion (French version), the default created category with a Horus profile is always in English.
I guess this one is a special one (common with the radio firmware).
[image]

LapinFou, 2017-03-27:
Similar issue: when importing models (_Mod**è**les_ in French) from a X series radio to the Horus, the foreign character is not converted.
[image]

3djc, 2017-03-27:
There should not be an accent there since radio can't display it. I guess a translation has an accent where it should not

LapinFou, 2017-03-27:
I guess so, but when the Companion translation is done, it is hard to know if it is a special case or not. Unless there is a way with Qt to display a programmer comment in the .ts file in order to warn the translators.

3djc, 2017-03-28:
I've just pushed a change with a warning for translators, you should be able to see it, let me know if this is not the case (this instance of models is modellist.cpp line 281 (280 before the change)

LapinFou, 2017-03-28:
I tried on the **next** branch, but it didn't work. See the generated .ts file
[image]

kilrah, 2017-03-28:
It probably won't work if you don't have the full environment setup. It needs to be able to find the source files and display the contents on the "Sources" panel..

[image]

3djc, 2017-03-28:
Wrong one:

[image]

mpaperno, 2017-03-28:
Those `QObject::tr`s should really be qualified with a namespace... but there are 438 of them...  😮

mpaperno, 2017-03-28:
Actually... `TreeModel` inherits from QObject so the `QObject::` qualifier should just be removed in this case... not sure why it's there.

LapinFou, 2017-03-28:
Indeed it was the wrong one (didn't realized at the same time you added the **Developer comments**, you fixed the issue 😓 )
[image]

PR #4712 (2017-03-29): Make default category translatable (with dev warning)
Thix fixes #4702
closed 2017-03-29 by mpaperno