opentx/opentx #4703: PL 2.2 Compilation error.

driver33, 2017-03-27:
I have faced following error when trying to pull firmware via CompanionTX 2.2:
"Compilation Error"
Options of firmware I made on : 
Taranis X9D
ppmus
eu
lua
massstorage
sqt5font

schwabe, 2017-03-27:
```cmake -DPCB=X9D -DPPM_UNIT=US -DLUA=YES -DUSB=MASSSTORAGE -DFONT=SQ5 -DSUPPORT_D16_EU_ONLY=YES ~/oss/opentx``` runs fine here on the 2.2RC14 tag.

driver33, 2017-03-27:
try Language=pl.

schwabe, 2017-03-27:
``` cmake -DPCB=X9D -DPPM_UNIT=US -DLUA=YES -DUSB=MASSSTORAGE -DFONT=SQ5 -DSUPPORT_D16_EU_ONLY=YES TRANSLATIONS=PL -DCMAKE_PREFIX_PATH=/usr/local/opt/qt5  ~/oss/opentx``` still works fine for me. Might be something to do with toolchain versions or someone who has access to the compilation server has to check the logs.

driver33, 2017-03-29:
Still unable to pull firmware night-builds PL version - setup of the firmware : all OFF; Language PL.

3djc, 2017-03-29:
And it will remain like that until next nightly

driver33, 2017-03-29:
Isn't it go directly from NEXT ?

3djc, 2017-03-29:
No, it will only when we reach release

driver33, 2017-03-29:
OK closing it then.

closed 2017-03-29 by driver33