opentx/opentx #4704: EEPROM Flashing problems

Phil1988, 2017-03-27:
I have problems Flashing my Turnigy 9X using an arduino Nano with a 100Ohm resistor between RST and 5V.
I can flash the OpenTX firmware but I always getting "EEPROM Warning" errors on flashing.
I can read/backup the EEPROM image but as I want to write it, Companion 2.2 (companion-windows-2.2.0N366) I get "Can't extract RADIO/radio.bin".

If I try it with Companion 2.1.9 I get an "Invalid Models and Settings file C:/.... " error.

How can I solve this problem?
Would be a real big problem if I have to set everything up from the scratch eacht time I flash a new firmware...

lezabour, 2017-06-06:
same problem here ;( did you find a solution?

Phil1988, 2017-06-06:
Sadly no... and I am still waiting for my USBasp flasher to arrive to check if it solves this problem.
But nice to hear I am not the only one with this problem.

Please keep me updated if you find a solution - so will I do!

Wesie, 2017-06-17:
For the 2nd Issue I'll believe, that is normal, because an Eeprom with the Modells are not downward compatible. This means, you can go only up, but never down (In this case its nicw to have a backup feom your models) 
The first one I couldn't see the one with N370. 
I've make a backup and restore them (9X Atmega 64A)
Maybe it's solved in the meantime?

Phil1988, 2017-06-17:
What do you mean with "downward compatible" ?
It is not possible to download any model settins - even after I install the latest firmware on the TX, setup a new model there and try to download this model/eeprom.

So I can always just "overwrite" the existing firmware and not do any backup!

I also have a 9X with an ATMEGA 64A.

kilrah, 2017-06-17:
Companion is most likely broken and doesn't take into account the fact that the 9x eeprom format hasn't been updated in a while, neither have the features. 
None of us has been using a 9x for years. We don't have any test hardware anymore so support for it is currently "best effort" with rather low investment.

Phil1988, 2017-06-17:
2.1.9 didnt worked as well... so what is the latest working version for the 9x?

Its really a pitty that this transmitter is not fully supported (anymore) because the transmitter itself and this software are great...

szalat, 2018-05-05:
Hello, I want to add my two cents here. Since I've read (very sparse info) about bug in new version affecting 9X (https://www.open-tx.org/2018/02/26/avr-pull) I was searching for more about it, found only this bug report.
I think it will be great to add some information to main site, e.g. just like kilrah wrote:

> (...)
> None of us has been using a 9x for years. We don't have any test hardware anymore so support for it is currently "best effort" with rather low investment.
> 
> You could jsut stay on 2.1, since there is nothing new at all in 2.2 for the 9x. Not even sure 2.1 had anything over 2.0.

I think it is vital information for 9X users searching for new firmware (following Companion reports: hi there, we have new version): 
"don't use new version of Companion nor Opentx. Just stick with version xxx."

Regards !

3djc, 2018-05-05:
I have reworded that post.

closed 2019-08-19 by kilrah