opentx/opentx #4706: Compilation 2.2 problem.

driver33, 2017-03-27:
I have some time now to fix PL translation for 2.2 version so I tried to setup compilation virtual machine on ubuntu 16.04.2.
I made fresh installation then I followed very strictly instruction on the page : https://github.com/opentx/opentx/wiki/OpenTX-2.2-Linux-Build-Instructions,

OK Results : 
1. cmake -DPCB=X9D+ -DGVARS=YES -DLUA=YES -DDEBUG=YES -DCMAKE_BUILD_TYPE=Debug ../
passes no errors or warnings,
2. make companion22  - passes no errors.

Not OK Resutls : 
1. make opentx-taranisplus-simulator - no rules to make object  'opentx-taranisplus-simulator'. Stop. 
But this is not a big issue - maybe tome misstype in wiki.

2. make firmware : 
  4%] Built target firmware_translations
Scanning dependencies of target taranis_xbm
[  4%] Generating sticks.lbm
Traceback (most recent call last):
  File "/home/jpd/opentx/opentx/radio/util/img2lbm.py", line 9, in <module>
    from PyQt4 import Qt, QtGui
ImportError: No module named PyQt4
radio/src/bitmaps/212x64/CMakeFiles/taranis_xbm.dir/build.make:60: recipe for target 'radio/src/bitmaps/212x64/sticks.lbm' failed
make[3]: *** [radio/src/bitmaps/212x64/sticks.lbm] Error 1
CMakeFiles/Makefile2:520: recipe for target 'radio/src/bitmaps/212x64/CMakeFiles/taranis_xbm.dir/all' failed
make[2]: *** [radio/src/bitmaps/212x64/CMakeFiles/taranis_xbm.dir/all] Error 2
CMakeFiles/Makefile2:155: recipe for target 'radio/src/CMakeFiles/firmware.dir/rule' failed
make[1]: *** [radio/src/CMakeFiles/firmware.dir/rule] Error 2

Where is the problem  ?

schwabe, 2017-03-27:
The instructions are missing pyqt. Could you try to do ```apt-get install python-pyqt5``` and see if that fixes the problem for you?

driver33, 2017-03-27:
Thx for hint; this do the trick. Fix in wWki is in need :)

schwabe, 2017-03-27:
Thanks for testing. I added python-pyqt5 to the wiki for 16.04

closed 2017-03-27 by schwabe