opentx/opentx #4707: Companion Simulator GVARs Not Being Read (Version 2.2.0N366)

RCdiy, 2017-03-27:
Companion > Edit Model > Flight Modes > FM0 
Any GVAR set a value
Simulate > View > Radio Outputs

mpaperno, 2017-03-28:
Can't reproduce in current `next` code.  Anyone else?  Might need more specifics... radio type, example model file, operating system...

RCdiy, 2017-03-28:
Apologies. 
Taranis Q X7 
Mac
I can make a video.

RCdiy, 2017-03-28:
I open Companion 
New models and settings 
Go through new model wizard 
Then steps from first post

3djc, 2017-03-28:
Can't replicate either, but when trying too I made a mistake, maybe you do the same : where di you input your gv value ? You should not use the large area (this is for name) but the small one with arrows

[image]

mpaperno, 2017-03-28:
Yea, still can't recreate on X7 simulator, even on Mac.... :-/

I made the same mistake with the name/value fields though.  My GVAR was named "10" and value was zero, which I found a bit peculiar for sure.

3djc, 2017-03-28:
Not urgent, but we could probably havve a look at the size of the field, it is a 'bit' large for the alowed 3 chars :D

kilrah, 2017-03-28:
Hopefully that helps... although it still does require people to read

RCdiy, 2017-03-28:
Apologies:) I do try and be thorough before opening an issue.
Works, but if I set GVAR1 in FM0 is it expected that GVAR1 in all flight modes have the same value?
Also is the simulator output going to later have the GVAR1 labels/names reflected?
[image]
[image]

projectkk2glider, 2017-03-28:
What you have there is that by default gvar value uses value from FM0 for all other flight modes. You can change that of course, but that is the default.

3djc, 2017-03-28:
Gvar has been given a column names already, available in next RC. Closing this as this is not an open issue anymore I think

closed 2017-03-28 by 3djc