opentx/opentx #4718: Can't play sound file with blue arrow in Companion 2.2

profdrdenis, 2017-03-29:
In companion special functions I select sound file, work well with simulate but I can't play with the blue arrow.

Thanks

3djc, 2017-03-30:
Companion version ? Os type ? Os version ?

profdrdenis, 2017-03-30:
Sorry,

now I use Companion 2.2.0N366 not work with first 2.2 edition. Vers. 2.19 work very well. 
My system:  Windows 10 Pro N, 64-Bit

mpaperno, 2017-05-06:
Is this still an issue?  Need more info if so, working OK here...

profdrdenis, 2017-05-06:
I use now N371 its the same. The last it was working was OpenTX 2.1.9. Which info you use?

mpaperno, 2017-05-06:
@kilrah  Might need to add some more plugins to the distro...  [windeployqt](http://doc.qt.io/qt-5/windows-deployment.html#the-windows-deployment-tool) recommends some I don't think we include now.  Specifically:
```
plugins/audio/qtaudio_windows.dll
plugins/mediaservice/qtmedia_audioengine.dll
```
It also recommends some other which we most likely don't ever need, so take that with a grain of salt.  In a debug session I don't see those being loaded, but I only checked on Win7 SP1 and don't have quick access to Win10 debug right now.  Plus who knows when it actually needs them or not...

@profdrdenis  Could you try extracting the [this zip file](http://max.wdg.us/OpenTX/qt5.7.1_msvc_audio_plugins.zip) into your Companion 2.2 installation folder? It should add a file to the already-existing `mediaservice` folder and add a new `audio` with one dll file.  Then restart Companion and see if audio works?  Thanks.

profdrdenis, 2017-05-06:
I install in my Companion 2.2 folder: not work.

Maybe different when I use simulator the play file(s) works

Waldo-P-Barnstormer, 2017-05-11:
Companion 2.2.N371
To add some more info it also no longer works on Win XP in Special Functions, but plays fine in the simulator. I tried the contents of the zip file, made no difference.  (In 2.1.9 it works in both places.)

On Win 7 it works in both places also, can't help with Win 10, no need to move there yet.
Jim

elecpower, 2017-05-14:
Can report the issue can occur on Ubuntu 16.04.

When click the play control the first time the following is reported in the OpenTX 2.2 debug widow 
[W] ::unknown():0 - defaultServiceProvider::requestService(): no service found for - "org.qt-project.qt.mediaplayer"

Unable to confirm if ever played on 2.2.0. Only just made a new sound file for a new model and tried to play it from Special Functions.

My test environment is a local compile of next on basic Ubuntu 16.04 64bit VM with OpenTX tool chain for Taranis X9D+

projectkk2glider, 2017-05-14:
Confirmed here also, Linux Mint 17.3. using `libqt5multimedia5:amd64                     5.2.1-0ubuntu5`

projectkk2glider, 2017-05-14:
Fixed here by installing `sudo apt-get install libqt5multimedia5-plugins`

mpaperno, 2017-05-14:
I still suspect Windows issue is also missing plugin DLL(s), but can't confirm.

kilrah, 2017-05-14:
I know that a couple of months ago when I did the last packaging fixes I tried it on a fresh Win VM and it worked, strange that it doesn't anymore...

Will have to do the same again, but won't be able to do so before next weekend

elecpower, 2017-05-14:
Retested successfully on Ubuntu 16.04 from updated package

nostromo123, 2017-05-15:
Debian Stretch - works "out of the box". PCLinuxOS is missing that lib. Not working.

Stromovik, 2017-06-05:
Same issue for me on Linux Mint 18.1
Solved by "sudo apt-get install libqt5multimedia5-plugins"

tmenet, 2018-01-28:
Win Pro N problem FIXED!!  I was having the same issue with windows Pro N 64bit with all the latest updates. I just updated Companion from 2,1.9 to 2.2.1 It was working fine earlier in 2.1.9  Upgraded to 2.2.1 amd computer did not play wave file selected in Model Special Functions Tab under parameters.  Wave files played just fine in simulator. I tried the zip file suggested earlier and it did not work.

Here is the fix. I noticed all that were having issues were using the Pro N version. The N version has no media support on install but Microsoft allows you to download a media pack for it. The problem is that every time they do a major update (like we had recently) You are required to update the media pack on your own specific to the new release.  As of right now this is the latest media pack https://www.microsoft.com/en-us/software-download/mediafeaturepack Otherwise search for Media Feature Pack for N versions of Windows 10 (your vindows version number)

Update your media feature pack and all works fine.

mpaperno, 2018-06-05:
The QtMultimedia plugins are included in the Debian package by CMake script. 

Also the required QtMm Windows plugins have been included in the installer package.

The original report was apparently due to lack of media support on Windows "N".

Good to close this?

kilrah, 2018-06-05:
Yup

closed 2018-06-05 by kilrah