opentx/opentx #4723: Lua find the error

merinkumar, 2017-03-30:
Hi Guys,
     Am new to lua and opentx, going thru the documentation for 2.2 and I have a Taranis X7 . I was able to run couple of scripts, but when i to test something, it errors, but the problem is the it dosent show the complete error, there is a flash second, when u can hardly see something written as syntax error. So my question is when you execute a script and get error, how to find the actual error code ?

below is the example i executed, but dint run, error...

` local function init_func()
  -- init_func is called once when model is loaded
end

local function bg_func()
  -- bg_func is called periodically (always, the screen visibility does not matter)
end

local function run_func(event)
  lcd.clear()
  lcd.drawText(1, 1, "drawChannel('Alt') example",0)
  lcd.drawText(1, 16, "using drawChannel(): ", 0)
  lcd.drawChannel(lcd.getLastPos()+20, 16 , "Alt", 0)
  lcd.drawText(1, 31, "using getValue(): ", 0)
  lcd.drawText(lcd.getLastPos() + 22, 31, getValue("Alt"), 0)
end

return { run=run_func, background=bg_func, init=init_func  } `

merinkumar, 2017-03-30:
sorry i found the debug ouput F6 in sim. And in there I found the below lines...

function returned unexpected value ..

then added return 0 to the run_func .. it worked.... I guess this is the methid to debug any error ?

3djc, 2017-03-30:
Indeed

closed 2017-03-30 by 3djc