opentx/opentx #4725: MULTI protocol not showing in External RF protocol 2.2.0N366 build

asmeureanu, 2017-03-31:
I have upgraded the Taranis X9D+ to opentx-x9d+ with appropriate SD card data however the
MULTI protocol not showing in External RF protocol list.

3djc, 2017-03-31:
Have you selected  multiprotocol in the options ?

asmeureanu, 2017-03-31:
Is there an extra option excluding the External RF protocol slot that is missing?

3djc, 2017-03-31:
[image]

asmeureanu, 2017-03-31:
this is in the companion, let me check and confirm

asmeureanu, 2017-03-31:
non issue had wrong non multi module fw flashed

closed 2017-03-31 by asmeureanu

lauchablanca, 2020-04-17:
Hola gente. Mi QX7 tenía la versión 2.3.0, pero no veía la opción MULTI Módulo.
Tuve que hacer un DOWNGRADE, a la versión 2.2.4, para poder elegir la opción MULTIMÓDULO en el Companion.
Por qué desapareción esa opción desde la versión 2.2.4 en adelante?
No se puede agregar a versiones posteriores a 2.3.0?

3djc, 2020-04-17:
English only please