opentx/opentx #4730: tranis x9E wireless trainer no longer works

bosshog99, 2017-03-31:
hello,

i've been using the X9E with a futaba (slave) with a DFT module installed.  the master (X9E) is using the aloft hobbies linker which has a D4R-II.  this setup worked perfectly with version 2.1.8.  after upgrading to 2.2.0N367 i can no longer use my trainer radio.  i have tried to rebind, create new models and nothing seems to work.  setting are on PPM as before.  it seems that the transmitters are binging fine but when i flip the momentary switch none of the values are passed between the two transmitters.  yes the switch is setup in the special functions menu, the protocol is PPM (timing set as before, default) in the master radio the trainer menu has been configured and the slave radio nothing has been changed and is configured correctly.

is anyone else having this issue?

3djc, 2017-04-10:
You mean this :
[image]

bosshog99, 2017-04-11:
yes, correct!  in the X9E the last option that shows on that screen from External RF is:
External RF
>MODE
>Bind on powerup
>Low power mode
>> then it scrolls back to the top <<  to Model name
there is no way to use the trainer function from the radio.

3djc, 2017-04-11:
Well that's taken from X9E. Here is one taken from a real radio, so it is there ....
[image]

kilrah, 2017-04-11:
> Bind on powerup
Low power mode

These are multiprotocol options that do not exist in official 2.1 OpenTX.  You're probably using the unofficial builds that include it and it's broken in those. Nothing we can do about it. Use OpenTX 2.2 instead, either now or after waiting for official release.

closed 2017-04-11 by kilrah

schwabe, 2017-04-11:
As a sidenote, even in newer 2.1.9 inofficial builds the problem should be fixed but as @kilrah said, if you want multiprotocol support use OpenTX 2.2

bosshog99, 2017-04-16:
guys, i am using 2.2.  ver. 2.2.0N367 exactly.  do you mean that i need to downgrade to ver. 2.2?  is the 0N367 part an unofficial build of 2.2?

sorry for my lack of understanding.

3djc, 2017-04-16:
As shown in the photo above, on N367 the option IS present. Please give us you OTX or more info